biancode/node-red-contrib-modbus #1: Modbus over serial

iurly, 2016-08-06:
Hi,
I understand node-red-contrib-modbus only supports modbus over TCP.
I need to deploy an application where connections are performed over a serial line.
I believe jsmodbus already supports serial connections, so the only missing part would be to add the node-red wrapper for it.
Is such a feature planned at all? If I were to implement this myself, would this be a welcome feature?
Any suggestions on how to proceed?
Thank you!

biancode, 2016-08-07:
There were no more time to develop that feature and nobody asked for it until now. I'll start to develop that.

iurly, 2016-08-15:
Hi,
I'm willing to contribute and I just started playing with it, though I am a newbie as far as github, node.js and node-RED are concerned.
Is there a quick way to get your feedback (in a gerrit-like workflow) without causing even more burden on your part?
Pushing draft changes to a forked github repo and issuing a pull request sounds a bit like overkill to me (especially at this stage where I'm not quite sure what I'm doing and it's very likely there's something wrong or missing...).
Thank you!

biancode, 2016-08-16:
Hi, the best way is to fork this repo, develop your part, and send a pull request to my repo from your forked repo. I'll do the review of your code and some tests and then we'll release if it works ;)

start here: http://nodered.org/docs/creating-nodes/

iurly, 2016-08-17:
Done! See pull request #2 . 
I guess I should've mentioned the issue in the commit message.
Feel free to shoot if there's something obviously wrong (no need to be polite...). :-)

biancode, 2016-08-19:
please pull again and test ...

biancode, 2016-08-20:
working well on a serial modbus software server

biancode, 2016-08-23:
see v0.6.0 and above

closed 2016-08-23 by biancode