biancode/node-red-contrib-modbus #3: Failing TCP connection causes the whole flow to apparently hang

iurly, 2016-08-17:
If I configure a TCP connection to a non-existing node or port, as soon as I deploy a flow, I end up with the following burst of messages on the console:

17 Aug 17:14:57 - [info] [modbus-client:ae8ea692.79b26] Client -> Connection 1 closed TCP at 127.0.0.1:502 unit_id: 1
17 Aug 17:14:57 - [error] [modbus-read:6f6640b8.e6c058] Error: connect ECONNREFUSED 127.0.0.1:502
...

without any timeouts in between.
This actually blocks my browser (firefox), too.
I believe further attempts should be performed at regular intervals, and not continuously.

biancode, 2016-08-18:
I'll check that, it should show "download ..?" under the node if the device is blocked. A delay time to connect would fix that.

biancode, 2016-08-20:
solved

biancode, 2016-08-20:
solved with the new reconnectTimeout parameter

biancode, 2016-08-23:
see v0.6.0 and above

closed 2016-08-23 by biancode