biancode/node-red-contrib-modbus #5: Interested to try modbus serial

k30r88, 2016-08-30:
Dear @biancode, I saw that you have branch : serial fro node-red-contrib-modbus 
Is it ready to try ? Also is this modbus RTU over serial or modbus ASCII over serial ? I am trying to connect a Modbus slave to BeagleBone that runs node-red (as a Master)

Martin

biancode, 2016-08-30:
Hi Martin,

give the new version 0.6.x a try. It is serial ready. The branch serial was the development branch from 0.5.x to 0.6.x master.

k30r88, 2016-08-30:
Thanks for your reply. I saw that from mikakaraila/node-red-contrib-modbus github, it is still version 0.5.1
I will give it a try from your biancode/node-red-contrib-modbus  version 0.6.3
Do you have plan to merge with mikakarraila/node-red-contrib-modbus ?

biancode, 2016-08-30:
Yes, pull request is done. Mika will test and merge it, but for now my repo is the leading repo for the development of node-red-contrib-modbus. Mika and me are in contact and we bring all changes back to his root repo.

biancode, 2016-08-30:
https://www.youtube.com/watch?v=h1usAIv5Pog

k30r88, 2016-08-30:
We looked at your youtube video, it demonstrates the modbus TCP not the modbus serial. We still have not gotten any luck with 0..6.3 using modbus serial

biancode, 2016-09-29:
please try version 0.7.6 - I should get a real device in the next few days to test it with hardware and not only software.

SuperJojo2001, 2016-10-08:
@biancode. Let me tell you how I test modbus RTU on my PC running Node-RED under windows.
1.)I have loaded a COM port emulator for Windows here: http://www.virtual-null-modem.com/. It creates a COM port #1 and a COM port #2 under Windows. Furthermore the tool emulates as if both ports are connected via "virtual" serial RS232 cable.
2.)As next I have loaded the Modbus RTU simualtor from here http://www.plcsimulator.org/

3.) I start now node-red with Modbus configured to serial and connect to COM port #1. Then I start the modbus simulator which connects to COM port 2. Since both ports are "virually connected", the telegrams from Node-RED go straight to the simulator and the responses back to it.

Works fine with node physical Modbus RTU device.

biancode, 2016-10-09:
Hi @SuperJojo2001,

Modbus PLC Simulator is already in use to test - the COM port emulator is cool, thank you!

others in use:
https://modbustool.codeplex.com/
http://www.modbusapps.com/mbp.php
http://www.chipkin.com/products/software/modbus-software/cas-modbus-rtu-parser/

more:
http://www.modbus.org/tech.php

biancode, 2016-10-09:
starting command Node-RED with serial debug:   DEBUG=SerialPort node-red -v

biancode, 2016-10-10:
@k30r88 please test version 0.8.x

closed 2016-10-10 by biancode