biancode/node-red-contrib-modbus #6: Using Modbus Read node in serial mode causes TypeError

SuperJojo2001, 2016-09-23:
I installed the latest version V0.7.5 today under my windows installed Node-RED.
Using the Modbus-Read node on COM1 produces the following error in Debug-Window:  TypeError: Invalid "baudRate" must be a number got: undefined

[image]

I cannot change "baudRate" other than just to make a selection in the drop down box as the picture shows you. Did I do anything wrong?

biancode, 2016-09-24:
I'll check this in the next few days.

hitosnap, 2016-09-29:
I am also facing same issue. I am using Node-red on Windows 7. it may be due to multiple serial library.

SuperJojo2001, 2016-09-29:
The error comes from the used "node-modbus" where in the source code "modbus-serial-client.js" there is a typo

serialport = new SerialPort(this.portName, {
                baudRate: this.baudrate,
                parity: this.parity,
                dataBits: this.dataBits,
                stopBits: this.stopBits
});

Must be  "baudRate: this.bauRrate," instead

biancode, 2016-09-29:
Oh, that's may an merge error.

biancode, 2016-09-29:
please try version 0.7.6

SuperJojo2001, 2016-10-08:
works, but I made a fork in node-modus solving all other problems too.

closed 2016-10-08 by SuperJojo2001

biancode, 2016-10-09:
@SuperJojo2001 my node-modus-1 is also forked from node-modbus to bring the name of NPM package in sync with the repo. The changes of GitHub jsmodbus (node-modbus) will be merged also in GitHub node-modbus. All improvements in serial testing will be pushed in node-modbus to be a bit faster with my node-red-contrib-modbus state of serial. Serial writing works for me well, but reading serial is a lot of work to be done. If it's finished it should be go back to jsmodbus repo.

SuperJojo2001, 2016-10-09:
Ok I understand.

 

In order to come further please watch my changes in the pull request of  https://github.com/Cloud-Automation/node-modbus in the file modbus-serial-client.js where I added also the correct CRC check routine and also the CRC verfication of a modbus response.

 

 

 

 

Gesendet: Sonntag, 09. Oktober 2016 um 09:44 Uhr