biancode/node-red-contrib-modbus #7: Modbus RTU slave address in serial stream constantly = 0x01

SuperJojo2001, 2016-09-23:
I suppose that the configuration parameter "Unit Id" is configuring the Modbus Slave destination address the nodes command shall be sent to.
If I am using now a Modbus Read node in Version 0.7.5. then whatever I am writing into "Unit Id" (default = 1) and deploy the flow, the serial network traffic analyser shows me e.g. modbus read holding request telegram such as 0x01 0x03 0x00 0x00 0x00 0x01 0x00 0x05 which is correct in the length, but indicates me as first byte(address) always the value 1. 
Where can I change the address the modus command shall be sent to?

biancode, 2016-09-24:
That should be also a problem of jsmodbus. Please open your issue on https://github.com/Cloud-Automation/node-modbus
There are only unitid settings used in tcp implementation for now.

biancode, 2016-09-29:
please try version 0.7.6 for that

SuperJojo2001, 2016-10-08:
@biancode, there are many more problems in Modbus serial "master" branch. I have recognized you made a fork meanwhile. I have made another pull request, which corrected also the problem of wrong CRC calculation for frame sending. Also the response CRC is checked there now and before the response is returned to the emitter, the first byte is trunkated which is the mirrored Modbus RTU slave address.

closed 2016-10-08 by SuperJojo2001