biancode/node-red-contrib-modbus #8: Wrong checksum in serial mode

SuperJojo2001, 2016-09-23:
When I am issuing a simple Read Holding register at addresss 0 with 1 register with node 0.7.5. I see the node pushing the stream 0x01 0x03 0x00 0x00 0x00 0x01 0x00 0x05 to the serial line.

The point is that the checksum at the end is sent as 0x00 0x05, but it must be 0x84 0x05. The first byte seem to be cut off.

SuperJojo2001, 2016-09-23:
sorry checksum must be 0x084 0x0A

SuperJojo2001, 2016-09-23:
watch spec here http://www.modbus.org/docs/Modbus_over_serial_line_V1_02.pdf

biancode, 2016-09-24:
Hi,

thanks for interest, but if you have problems with modbus, you should contact the jsmodbus developer https://github.com/Cloud-Automation/node-modbus 

This package is based on jsmodbus/node-modbus and it just parameterizes and uses that modbus package. The node-red nodes are controls for jsmodbus.

closed 2016-09-24 by biancode