biancode/node-red-contrib-modbus #9: Missing Modbus float registers capability

tridiumcontrols, 2016-09-24:
Hi, I have installed http://flows.nodered.org/node/node-red-contrib-modbus and I cannot find an area to change the value type for modbus. Seems like the node is a somewhat complete, but is missing few things. Primarily the value type. There are signed and unsigned integers as well as floats. The Modbus node is able to talk to unsigned integers just fine, but I have a current sensing energy monitoring CT's that use floats as their registers. I assume I need to use a function inorder to achieve what I need to do, but I just today i found out about node-red and learning the ropes. Can someome assist in the function or what needs to happen to get the value I am looking for. 

I'm using modbus TCP. 

Here's the CT documentation 

https://www.ccontrols.com/support/dp/EnerceptMeterInstallationGuide.pdf starts on page 17 for the addresses. 

THank you.

biancode, 2016-09-24:
That shouldn't be a feature of the nodes. It isn't a good idea to do one for all in the node, if you think about reading different values (Bool, Int8, Int16, Int32, Int64, all the unsigned ...). The nodes give just the raw data and they are 16Bit chunks in Modbus. Node-Red is to have very modular nodes. Bitshift or bit-shift is the keyword to search for a solution. With that you can connect the 16Bit chunks in a function node. You can take a look here: http://bit.ly/1OzLXxz

closed 2016-09-24 by biancode

biancode, 2016-09-30:
https://www.youtube.com/watch?v=OvRD2UvrHjE