biancode/node-red-contrib-modbus #10: Feature request for dynamic Modbus frame setup

SuperJojo2001, 2016-10-09:
@biancode. I like the "Getter node" very much to trigger requests on demand.

I want to explain you where I see a big improvement especially when using it with Modbus serial on RS485 interfaces. A RS485 interface has the big advantage to allow to connect the full range of 255 Modbus slaves at the same time.

But today the node configuration of a Modbus serial connection is static, means it is bound to a single COM port with a setting of baudrate etc. but also the unit ID and register, offset and function code as constants. This allows the connection to one single Modbus slave only and a single register. While on Modbus TCP it works fine to define multiple config nodes to read/write different registers with different IP addresses ... on Modbus serial it doesn't work any more.

The point is that once the COM port is already in use you cannot use it for another configuration node with a different unitID or register or FC or offset setup. So you are not able to connect to multiple Modbus slaves on a single RS485 COM port interface.

This is why I have the idea to feed the "Getter node" not only with an empty triggering msg, but with a msg.payload that contains also the FC to be used, the register address, the quantity and finally unitID to allow dynamic frame setup on each request.

This is how a triggering message could look like

msg.payload.unitid
msg.payload.fc
msg.payload.quantity
msg.payload.register
msg.payload.offset

Now whenever a user sets up such a triggering msg instead of an empty one, the "Getter node" uses this msg.payload parameters instead of those in the config node. If the msg is empty it continues to work like before. Of course the injected response of the "Getter node" should mirror the msg.payload in the output in order to allow the user to distinguish the different responses.

With this method you are able to read/write from multiple modbus slaves with a single "Getter node".

biancode, 2016-10-09:
Hi,

may it is enough to have an option to tell the Getter to open before and close after every read and so you can use multiple Getter nodes?

SuperJojo2001, 2016-10-09:
The question is how node-red would handle multiple getters. 

Supposing we have in simplest case two getters active on the same COM port and the getters would now open and close the connection as you propose. I think if both getters are triggered at the same time by an inject node then node-red would call the two getters the same time. While the first getter opens the connection...sends data and waits for the modbus response for sure the other one is triggered also ... and then it wants to open the already opened port and gets an COM port open error. So I think this will not work.
I still see the only solution for sending mutliple message across one COM port is to do it with a single getter only and a msg that defines what to be sent

SuperJojo2001, 2016-10-09:
What might be possible to daisy chain multiple getters. This means starting in the flow with the first getter and connect its response output to the next second getter and from there to the third and so on. With this construction to read 100 different registers for example you need to insert 100 getters in your flow.

biancode, 2016-10-09:
Yes, for now the client opens the COM port once and is in an auto reconnect mode. What means if you become able to choose between auto connecting and open/close on read it should work, because of the single thread of nodejs. (is not possible for now) The client resource is only once for all getter or read/write nodes in node-red if you setup only one client.

There are two ways to handle multiple servers (slaves), if I'll enhance the Getter:

1. one event trigger to multiple Getters with one client (master)

[image]

2. multiple event triggers to one Getter with one client (master) with parameters how to connect to its servers (slaves)

[image]

In both cases, you can use the response and the data to write a function behind to know what to do with the data and where it comes from. What's your favorite?

SuperJojo2001, 2016-10-09:
Well if I have the choice now I would prefer the solution 1. The nice thing in sol 1 is that a user could give each getter node a specific name like "Read Register 0", the second "Read Register 1000" and so on base on what the getter node is really configured for. This makes the flow better readable. In sol 2 you can give the single getter node a single symbolic name only while all parameters are somewhere hidden in the background.

biancode, 2016-10-09:
So we agree with each other, but it takes time. I can do that in the next few days.

SuperJojo2001, 2016-10-09:
I have made a simple test to take control of the getter with msg.payload parameters
My getter node is fed by the following function node:
msg.payload = {};
msg.payload.fc = "HoldingRegister";
msg.payload.unitid = 1;
msg.payload.address = 0;
msg.payload.quantity = 1;
return msg;

The modbus-getter.js I have extended by some lines of code to overwrite configured values with those coming from msg.payload

        node.on("input", function (msg) {
			try {
			   node.dataType = msg.payload.fc;
			} catch(err) {};
			try {
			   node.connection.unitId = msg.payload.unitid;
			} catch(err) {};
			try {
			   node.adr = msg.payload.address;
			} catch(err) {};
			try {
			   node.quantity = msg.payload.quantity;
			} catch(err) {};
		...
	
works perfect now. For each message injected I can now change all parameters of Modbus on demand.

biancode, 2016-10-09:
for your CRC checkout node-modbus 2.1.0 and node-red-contrib-modbus 0.8.0

Now issue - split of PDU doesn't work - if your device didn't split it should work with serial RTU
[image]
 work:

DEBUG : Data flushed.
DEBUG : PDU with CRC written to serial
DEBUG : received data {"type":"Buffer","data":[1,6]}
DEBUG : received data {"type":"Buffer","data":[195,80]}
DEBUG : received data {"type":"Buffer","data":[0,17,117]}
DEBUG : received data {"type":"Buffer","data":[147]}
ERROR : Request timed out after 5 sec

SuperJojo2001, 2016-10-09:
yes this is correct. This routine does not cover splitted APDUs till now and treats them always as a "whole". In case of splitted APDUs a wrapper is needed to concat each of the single strings similar to the str.concat() function

Just a silly question maybe...but I have never seen Modbus RTU slave sending splitted streams in my whole life so far? Or is it coming by the very nature of node-red to split streams from time to time?

biancode, 2016-10-09:
I,ve a neru Prizm230 and that device starts splitting after some time - there is no setup option for the prizm to control splitting. It serial over USB, may it is a problem of that driver.

SuperJojo2001, 2016-10-09:
Interesting to hear that. It might be the USB to serial driver causing this problem.
But I read a little further in the Modbus RTU specification. http://www.modbus.org/docs/Modbus_over_serial_line_V1.pdf
It says on page 13, that "If  a  silent  interval  of  more  than  1.5  character  times  occurs  between  two  characters,  the  message  frame  is  declared  incomplete  and should be discarded by the receiver"

So if the serial port is configure to a timeout value of 1.5 char times to detect the end of frame we are fine. If it is more - which may be the case with your device - the response needs to be ignored and discarded.

SuperJojo2001, 2016-10-09:
I checked node-modbus 2.1.0 now and it works perfect right away. super!

Just a silly question but what is this big/little endian thing check made for in the onSend() function? Is is to check on which processor the source is running on?

biancode, 2016-10-09:
https://en.wikipedia.org/wiki/Endianness - the PRIZM230 using LITTLE and the most of jsmodbus is BIG

SuperJojo2001, 2016-10-09:
The thing with the endianess astonishes me since I see no necessity to check the endianess at any time.

I programmed the modbus serial RTU stack in our company 21 years ago in "C" and since then we have never changed it any more. All modbus RTU slave devices I have ever met work in accodance to the spec mentioned in two statements above.

Read chapter 6.2.2 "CRC generation" there it says: CRC is ALWAYS placed in the LITTLE endian format in a Modbus RTU message. So a value 0x1241 for example will be place low byte first as 0x41 and then 0x12. 

So your PRIZM slave device works fine, if it placeing the CRC as the example. Please explain to me a little closer what "most of jsmodbus is BIG" means?

biancode, 2016-10-09:
Example: 
pkt = pkt.word16**be**(crc).buffer();  jsmodbus for now writes every time BIG-ENDIAN

if you need LITTLE you need: pkt.word16**le**(crc).buffer();

SuperJojo2001, 2016-10-10:
Ok a concrete example with real values: supposing the "crc.crc16modbus(buf)" caculation results in a decimal value 1000.  This value equals to 0x03E8 hexadecimal. Since Modbus RTU serial stream always follows the Intel format which is LSB first the CRC appended to the PDU must be ... 0xE8 0x03 in any case. So the code pkt.word16le(crc).buffer() will do it fine. The code pkt.word16be(crc).buffer() will do it wrong in any case and does never follow the Modbus RTU spec. This is why I am wondering why we should offer this "BIG" opion anyways. Look this is the example in the Modbus spec:, always low byte is coming first:
[image]

biancode, 2016-10-12:
[{"id":"56658acc.106c74","type":"modbus-flex-getter","z":"c2307dfb.e5bc9","name":"","server":"813bbf0e.dc574","x":520,"y":140,"wires":[["67272564.6cb83c"],["5675eeb9.d9487"]]},{"id":"5675eeb9.d9487","type":"modbus-response","z":"c2307dfb.e5bc9","name":"","registerShowMax":20,"x":770,"y":180,"wires":[]},{"id":"67272564.6cb83c","type":"debug","z":"c2307dfb.e5bc9","name":"","active":true,"console":"false","complete":"true","x":730,"y":120,"wires":[]},{"id":"fc2e0cd3.1a53f","type":"inject","z":"c2307dfb.e5bc9","name":"","topic":"","payload":"","payloadType":"date","repeat":"","crontab":"","once":false,"x":160,"y":140,"wires":[["c178914.e96fc7"]]},{"id":"c178914.e96fc7","type":"function","z":"c2307dfb.e5bc9","name":"","func":"var address = 0;\nvar input = msg.payload;\nfor(var item = 0; item < 4; item++) {\n    address = 20 + item;\n    msg.payload = { payload: input, 'fc': 3, 'unitid': 1, 'address': address , 'quantity': 1 };\n    node.send(msg);\n}","outputs":1,"noerr":0,"x":310,"y":140,"wires":[["56658acc.106c74"]]},{"id":"813bbf0e.dc574","type":"modbus-client","z":"c2307dfb.e5bc9","name":"","clienttype":"tcp","tcpHost":"192.168.1.250","tcpPort":"502","serialPort":"/dev/ttyS0","serialBaudrate":"9600","serialDatabits":"8","serialStopbits":"1","serialParity":"none","unit_id":"1","clientTimeout":"5000","reconnectTimeout":"5000"}]

biancode, 2016-10-12:
Please, try the new Flex Getter in v0.8.2 - known issue: the address is wrong on multiple single requests.

SuperJojo2001, 2016-10-13:
hello @biancode. I does exactly what it should and is perfect for flexible use.

I tested your script and got 5 messages as response. This node is now usable for us.
[image]

SuperJojo2001, 2016-10-13:
I have one comment. If you configure a serial node and keep for example the endianess drop down menu empty like this picture show (cause there is no default and you click ok my mistake):
[image]

It works fine, but if you want to open the config node again it will not open any more.

I would recommend to make a default proposal for users: Serial mode = RTU (99.99% of customer have this type of device) and Endian = LITTLE.

biancode, 2016-11-27:
should be fixed with v0.9.0 and above

closed 2016-11-27 by biancode