biancode/node-red-contrib-modbus #12: Writing one Register with "Preset Mutliple Registers" rejected

SuperJojo2001, 2016-10-10:
@biancode, the Modbus specification defines that the FC16 also supports writing one single register.

The spec http://www.modbus.org/docs/Modbus_Application_Protocol_V1_1b.pdf says in chapter 6.12  

16 (0x10) Write Multiple registers 
This  function  code  is  used  to  write  a  block  of  contiguous  registers  (1  to  123  registers)  in  a  
remote device.  

Today the write node rejects a quantity of 1. This should be corrected.

biancode, 2016-10-10:
If you want, please contribute on https://github.com/biancode/node-modbus-1 and we can bring it to the nodes and later back to jsmodbus (node-modbus)

SuperJojo2001, 2016-10-10:
@biancode please help me to understand payload handling of FC1 = preset multiple registers.
The msg.payload for FC16 must be an array of data? Means for writing 2 registers for example I have to set up msg.payload = [1,2]; for example?

biancode, 2016-10-10:
Yes, it could be an array of UInt16's or a buffer of I mean UInt8's/Bytes.

see WriteMultipleRegisters.js in jsmodbus:

_this.writeMultipleRegisters function (startAddress, register)_ 
register have to be an instance of **Buffer or Array**

If your question goes deeper in function you have to ask the author of jsmodbus "Cloud Automation" , please.

I'm doing this all in my free time, so I had no time to compare the whole jsmodbus/node-modbus with Sepcs for now. ;)
I've compared some I need with Sepcs and it wasn't correct in all cases and so I started to fix it step by step.

SuperJojo2001, 2016-10-10:
@biancode, yes I know you all do it free time. Me too now, but I want to get the node working properly.

Today in the morning I start testing the WriteMultipleRegister as you know and now I am back home and able to debug the code. 

I went indeed into this function

_this.writeMultipleRegisters function (startAddress, register)_

This function continues with following code:

_if(register instanceof Buffer) {
 ... 
} else if(register instanceof Array) {
 ...
}_

But now comes the problem: _register_ which is handed over from your code as msg.payload even when you set it to values like [1,2,3,4,5...] which is an Array() is always of type _object_ and never of Array or buffer etc. So all type checks fail in this.writeMultipleRegisters and no values are written at all.

For simple test entering this code in a function node and watch the debug window:
_msg.payload = [100,100];
node.warn(typeof msg.payload);
_

SuperJojo2001, 2016-10-10:
All websites I checked now to test if a javascript object is of type Array() is to use the function _Array.isArray()_ instead of _instanceof_
I verified it now in the source code and after modify WriteMultipleRegisters works fine now

biancode, 2016-10-10:
done with v0.8.1 and node-modbus 2.1.1

closed 2016-10-10 by biancode