biancode/node-red-contrib-modbus #17: COM port may not close fast enough under windows after deploy

SuperJojo2001, 2016-10-12:
This issue addresses a problem that windows does not close a serial port fast enough when you close it: find a comment in the remark field here https://msdn.microsoft.com/en-us/system.io.ports.serialport.close.aspx

If you run now node-red-contrib-modbus in serial mode under windows you are faced with the problem that once you pressed deploy sometimes the serialport module reports "COM port X : Access denied" even if it has issued a close before when the deploying process is started. So the time between closing and starting the flow is too short during deploy process.

This is why I checked the problem is gone if the init() call of modus-getter, read,write is delayed by a timer. In my case 100msec was enough

if(process.platform === "win32") {
           setTimeout(init,100); // last deploy on windows may not have closed the COM port so wait a little
} else {
          init();
}

biancode, 2016-10-12:
That should be fixed in node-modbus or in client node

biancode, 2016-12-04:
please test with v0.9.1

closed 2016-12-21 by biancode