biancode/node-red-contrib-modbus #19: Modbus RTU serial configured BIG endian sends wrong checksum

SuperJojo2001, 2016-10-13:
I have proven that sending the CRC checksum in BiG endian format the devices addressed ignore request frame. CRC has to be sent always little endian.
[image]

This screen shot show the data logging monitor of Rx and Tx line. First message you see correct checksum. Device replies. 2nd try was with setting BIG endian, device ignores frame and does not send back Tx frame.

modbus-serial-client needs to send CRC always in little endian format.

biancode, 2016-10-13:
Please, post some of the logs _"-ENDIAN with CRC via"_

SuperJojo2001, 2016-10-13:
First of all this is the log file of the modbus slave tool:
[Communication log.txt](https://github.com/biancode/node-red-contrib-modbus/files/528171/Communication.log.txt)
I took the flow you sent me yesterday to test modbus flex getter with it. Your upper button sends out reading address 20,21,22,23. This works pefect. 
Then the log continues with my next try while I set to serial port to BIG endian. Now you see the modbus slave tool receives the same message as the first request but word CRC is byte swapped.

This this is the debug log of node. It ends with timeout cause modbus slave receieves big endian CRC which is wrong.
[debug log.txt](https://github.com/biancode/node-red-contrib-modbus/files/528182/debug.log.txt)

biancode, 2016-11-27:
we are working now on a very Spec compliant modbus lib for the nodes

closed 2016-11-27 by biancode