biancode/node-red-contrib-modbus #21: Read from Modbus Master

Sineos, 2016-10-30:
Please forgive my noobish question, I'm new to dealing with modbus. I have a device, that works as modbus master only. I want to have node-red connect to this master as slave and receive readings at the rate in which they are send.
For each channel of readings in the master I can set:
- IP address of the slave
- Device ID of the slave
- Function ID
- Address ID
Whatever I tried I cannot get readings. It all ends up in an ETIMEDOUT state.

Any pointers would be highly appreciated.

biancode, 2016-10-30:
You can try out the server (slave) node.  The reader and getter are clients (master). Setup a server node, inject every sec and watch the output of the server node. Let me know how it works for you.

Sineos, 2016-11-01:
Thank you for this pointer.

I'm currently trying around with https://sourceforge.net/projects/qmodmaster/ to simulate a ModBus Master. While it seems to connect to the server node, I cannot get any data from it.
What purpose does the input pin has on the server node?
Is my understanding correct, that any data received from the master's first 1024 addresses of FC01 (Coil), FC03 (Holding) and FC04 (Input) should be msg'ed via the output pins?

biancode, 2016-11-27:
The "Modbus Server" node has 1024 crates for every block - 1024 Coils, 1024 Holdings, and 1024 Inputs - so you didn't get the first 1024 - you get all of them at the outputs to read their data. The data you will get are 1024 x 16Bit chunks you can read from.

closed 2016-12-04 by biancode