biancode/node-red-contrib-modbus #25: Modbus RTU read error

thegunshow, 2016-12-08:
Hi

this is the error I receive "Modbus Read client: {"err":"timeout"}". The USB RS485 device is installed and working on the raspberry.  I can get coms to a slave using another Modbus node but this one keeps failing.

Node-red v0.15.2
Node v0.10.29

Are there any dependencies that need to be installed?

Thanks
Brendan

biancode, 2016-12-08:
which version of the node package, please ;)

mmaehler, 2016-12-10:
Hi,
I have the same problem and installd is Version 0.9.4.
Node 4.7.0
NPM 2.15.11
Thanks in advanced!
Regards,
Marko

biancode, 2016-12-21:
please try v0.10.7 with Node 4.7.0 or 6.9.2 LTS

biancode, 2016-12-21:
or test the tagged pre-release node-red-contrib-modbus@0.11.1 on 6.9.2 LTS

mtoko, 2016-12-26:
Hi

Please could you advise on the versions of Node and Node Red that have been tested. I have Node v7.3.0 and Node Red v0.15.2 but cannot connect via RTU. I have used "node-red-contrib-serial-modbus" to test the serial connection and can communicate with the serial device but the output is a buffer. I have used the TCP version and this works great so would like to continue using this in RTU. The serial port config that works is /dev/ttyUSB0

Any advice would be appreciated

Regards
Darryl

biancode, 2016-12-26:
try npm install node-red-contrib-modbus@0.11.3

mtoko, 2016-12-27:
Thank you for the response
I have tried Node 6.9.2 and Node red 0.15 and node-red-contrib-modbus@0.11.3. There is an error

Error: Callback was already called 

and sometimes and additional error
Error: Port is not open

The flow below still works so I don't think this is a hardware problem

[{"id":"1b589266.659d3e","type":"modbusSerial in","z":"fb5470eb.ad5d8","port":"80c93e1d.bbc5f","slaves":"33","start":"2","count":1,"dtype":"coil","period":60,"topic":"","name":"","x":177,"y":70,"wires":[["1e6110ff.7ed07f"]]},{"id":"80c93e1d.bbc5f","type":"modbusSerialConfig","z":"","port":"/dev/ttyUSB0","baud":"9600","data":"8","parity":"none","stop":"1","name":""}]

Any advice would be appreciated

Darryl

biancode, 2016-12-31:
please try npm install node-red-contrib-modbus@0.11.7

mtoko, 2017-01-01:
Thank you for you help
The problem appears to be related to the version of Node js as you suggested. I have some basic communications working

Node js version 6.9.2
npm 3.10.9
Node red 0.15.2

The serial port config in my installation was /dev/ttyUSB0

mmaehler, 2017-01-02:
Many thanks for a lot of new versions! I wish you a happy new year!!!
After install from version 0.11.10 the /dev/ttyUSB0 works normally, sends and receives an answer. The modbus gateway signals interference-free communication.

But I unfortunately do not receive the desired data but the following error message  at any request:

2 Jan 20:21:40 - [error] [modbus-read:e7415af6.8ec4a8] Data length error, expected 7 got 1

Please help me!
Thanks in advanced
Regards
Marko

biancode, 2017-01-02:
Length 1 looks like a broadcast from modbus master - is your gateway a modbus master or slave?

presuntoufu, 2017-01-02:
mmaehler, i had the same problem. In my case, changing from RTU to RTU buffered solves the problem.

biancode, 2017-01-02:
oh yes, RTU needs response at once - buffered is to handle split responding

mmaehler, 2017-01-03:
Hi, thanks for the quick reply! 
With the setting RTU buffered I get data. Unfortunately, it is not the data I expect.

I use a WattNode® Modbus WNC-3Y-400-MB current meter with a user manual [https://ctlsys.com/wp-content/uploads/2016/10/WNC-Modbus-Manual-V18.pdf](url)

If I reading the integer frequency register at 1221. This register retrieves the frequency and must supply a value of approx. 500.
 I get the following error message:
3 Jan 22:22:31 - [error] [modbus-read:e7415af6.8ec4a8] Modbus exception 2

To limit the problem, I have tried an additional program. The name is Modpoll in version 3.4 [http://www.modbusdriver.com/modpoll.html](url)
With the parameters "modpoll.exe -a 2 -m rtu -b 9600 -d 8 -s 1 -p none -r 1221 -t 4 COM4" I get:
[1221]: 500
This is the expected result. I guess it's the setting:
-t 4          16-bit output (holding) register data type

With the setting "-t 4:float    32-bit float data type in output (holding) register table"  I get the following error message with modepoll.exe: "Illegal Data Address exception response!"

Can you please help?
Thanks
Marko

biancode, 2017-01-05:
did you try address 1220 or 1221 with quantity 1 - FC3 or FC4? Your LED should show green/red see Page28 in pdf file - what does it show?

mmaehler, 2017-01-05:
I have tested it with register 1221, quantity 1, FC4, RTU buffered and unit-id 2.

biancode, 2017-01-05:
It is a16 bit query. For 32 Bit you need quantity 2. Please try FC3, if you didn't. It should be the same for your device, but sometimes it helps. And the device has unit id configured with 2 ?

mmaehler, 2017-01-05:
I've checked every configuration. Among other things FC3 and FC4 with queries from one register or two registers. Also, I have queried different registers. When querying register 1707 I got the value 0. For queries from other registers I get data, which according to the register description would have to supply a different value.

The Modbus device is configured with 2 and also delivers correct data to my SolarEdge inverter. This is, of course, taken out of service during the tests, since there is only one master on theModbus bus.

biancode, 2017-01-08:
Modbus exception 2 - Illegal Data Address - so you're not able to read that address

please try:  
FC 4  Address: 1220
FC 3  Address: 1220

input registers
FC 4  Address: 41220
FC 4  Address: 41221

holding registers
FC 3  Address: 31220
FC 3  Address: 31221

Manual cite:

Modbus Register Addressing
There are a few points about Modbus register addressing that can cause confusion.
● In the Modbus specification, register numbers are documented as “one based”, but transmitted
as “zero based”. For example, we document that EnergySum appears at address 1001.
If you are using any Modbus software or Modbus aware device, you should use “1001” as the
register address. However, if you are writing your own low-level Modbus driver or firmware,
**you will need to subtract one from the register number** when creating the Modbus frame (or 
Operating Instructions 33
packet), so the actual register number that appears on the RS-485 bus will be “1000” (or in
hexadecimal, 0x03E8).
● A common Modbus convention adds the function code as a leading digit on the register
number, so a register like EnergySum (1001) would be documented as 41001. All input registers
(function code 04) would use the form **4xxxx**, while holding registers (function code 03)
would use the form **3xxxx**. The WattNode meter treats holding registers and input registers
interchangeably, and does not use this convention. If your Modbus software expects a leading
“3” or “4”, either will work for most registers, and use “3” for configuration registers.

closed 2017-01-13 by biancode