biancode/node-red-contrib-modbus #27: Incorrect handling of the response error messages from slave-device

ykornilov, 2016-12-13:
Hi

Wrong slave-ID  leads to collapse node-red

Best regards,
Yury

```
23 Nov 17:16:28 - RangeError: index out of range
   at checkOffset (buffer.js:506:11)
   at Buffer.readUInt16BE (buffer.js:560:5)
   at Object.<anonymous> (C:\Users\Y\AppData\Roaming\npm\node_modules\node-red-contrib-modbus\node_modules\node-modbus\node-modbus\modbus-tcp-client.js:104:36)
   at Socket.emit (events.js:107:17)
   at readableAddChunk (_stream_readable.js:163:16)
   at Socket.Readable.push (_stream_readable.js:126:10)
   at TCP.onread (net.js:538:20)
```

biancode, 2016-12-21:
please try v0.10.7

biancode, 2016-12-26:
try npm install node-red-contrib-modbus@0.11.3 on node 6.9.2 LTS and node-red 0.15.2 - v7 of nodejs is not supported from some node packages

biancode, 2016-12-31:
please try node-red-contrib-modbus@0.11.7

closed 2017-01-05 by biancode