biancode/node-red-contrib-modbus #28: Incorrect error handling in `set_modbus_error`

nallenscott, 2016-12-14:
Catch nodes are not being triggered when an error occurs during a read/write. I think this is because the original `msg` is not included in the call to node.error(). See the node-red docs on [error handling](https://nodered.org/docs/writing-functions#handling-errors). As a test, I edited [this line](https://github.com/biancode/node-red-contrib-modbus/blob/master/src/modbus-flex-getter.js#L264) in `modbus-flex-getter.js` for each of the switch cases:

```js
}).fail( function (err) { set_modbus_error(err, msg); });
```
... and [this line](https://github.com/biancode/node-red-contrib-modbus/blob/master/src/modbus-flex-getter.js#L357) in `set_modbus_error` to include the original `msg` object:

```js
node.error('Modbus Flex Getter client: ' + JSON.stringify(err), msg);
```

... and now catch nodes are working as expected :)

biancode, 2016-12-20:
API docs (http://nodered.org/docs/creating-nodes/node-js) says:

this.log("Something happened");
this.warn("Something happened you should know about");
this.error("Oh no, something bad happened");

biancode, 2016-12-20:
OK

biancode, 2016-12-20:
comming soon with v0.10.x

closed 2016-12-20 by biancode

nallenscott, 2016-12-20:
Why was this closed? Was it fixed?

biancode, 2016-12-20:
comming soon with v0.10.x - you can give it a try with the tag @0.10.5 on npm install

nallenscott, 2016-12-20:
Got it. Thanks!

biancode, 2016-12-22:
please try @0.11.2 with whole new serial engine