biancode/node-red-contrib-modbus #29: modbus-write and modbus-flex-write not working in 0.10.7

cfatfc, 2016-12-28:
FC16 works in 0.9.5, but not in 0.10.7. In 0.10.7, the node connects to the remote Modbus-TCP device, but there is no output when the flow is triggered.

biancode, 2016-12-29:
please try npm install node-red-contrib-modbus@0.11.4

biancode, 2016-12-30:
please try npm install node-red-contrib-modbus@0.11.5

biancode, 2016-12-31:
there is a whole new modbus communication and queueing in v0.11.x to get more control from the nodes in node-red - may you need more than one connection to your gateway, if the workload is to high - when all nodes sending together every second for example - or you have to find the right queue command delay for your device

biancode, 2016-12-31:
[image]

closed 2017-01-02 by biancode