biancode/node-red-contrib-modbus #30: FC6 not working in modbus-write

cfatfc, 2016-12-28:
version 0.9.5 modbus-write node returns the following error:

Modbus Write client: {"errorCode":134,"exceptionCode":11,"message":"GATEWAY TARGET DEVICE FAILED TO RESPOND"}

Can successfully write to a single holding register in version 0.9.5 with FC16.

biancode, 2016-12-29:
please try npm install node-red-contrib-modbus@0.11.4

biancode, 2016-12-30:
please try npm install node-red-contrib-modbus@0.11.5

biancode, 2016-12-31:
in the new version 0.11.7 you have to try setup Queue commands with true and play with the queue command delay time
[image]

biancode, 2016-12-31:
My testing flow with unbufferd and buffered VServer ( PLC Sim on Windows ) and some hardware (Schneider PLC TCP and Renu Panel Modbus Serial)

[image]

[image]

biancode, 2016-12-31:
Serial Modbus Panel
[image]

closed 2017-01-02 by biancode