biancode/node-red-contrib-modbus #32: Problem with unit_id

presuntoufu, 2017-01-02:
Hi biancode,
I'm having an issue with unit_id. My installation is node-red v0.15.2, node v4.7.0, npm v2.15.11 and node-red-contrib-modbus v0.11.10, everything in a raspberry pi B+. The problem is that my raspi only communicates with default slave id, configured on modbus-client. When i configure in a modbus-getter or a modbus-read, the unit-id still the same as default, never changing. Any tips?

biancode, 2017-01-02:
is a bug in HTML of Getter and Read - you can not save it in the Getter and Read, right?

biancode, 2017-01-02:
try v0.11.11

biancode, 2017-01-03:
see also my [modbus-serial issue](https://github.com/yaacov/node-modbus-serial/issues/40)

presuntoufu, 2017-01-03:
biancode, 
I tested v0.11.11, and modbus-read and modbus-getter now works as expected. Now the same problem occurs with modbus-write. Can you do your magic again??

thanks a lot

presuntoufu, 2017-01-03:
Just to add more information in this issue, when i exec modbus-write, i'm receiveing a "modbus exception 3" error. In my opinion, without the correct unit_id, the calculated crc is wrong, and the node throw this error.

biancode, 2017-01-03:
crc is an issue for modbus-serial, please

presuntoufu, 2017-01-03:
sorry, i'm just telling you what happens here, just trying to help in debugging.

biancode, 2017-01-03:
magic goes to v0.11.12

biancode, 2017-01-03:
can you see your unitid in console with node-red -v  (verbose mode)

presuntoufu, 2017-01-03:
I changed node-config-input-unitid for node-input-unitid, in modbus-write.html. Apparently, this saves the unitid when send data. But the "modbus exception 3" error still occurs. My unitid is 2 (in modbus-write) and default unitid is 1 (in modbus-config). 

3 Jan 13:47:50 - [info] [modbus-client:usb-485] Client -> {"info":"queue write msg","msg":{"_msgid":"80f26e29.7f0d9","topic":"","payload":{"value":true,"unitid":"2","fc":5,"address":100,"quantity":"1"}},"state":"CONNECTED"} Serial@/dev/ttyUSB0:9600bit/s unit_id: 1

3 Jan 13:47:50 - [info] [modbus-client:usb-485] {"loopyLength":0,"commandData":{"msg":{"_msgid":"80f26e29.7f0d9","topic":"","payload":{"value":true,"unitid":"2","fc":5,"address":100,"quantity":"1"}}},"delay":100}

3 Jan 13:47:50 - [info] [modbus-client:usb-485] Client -> {"info":"write msg","msg":{"_msgid":"80f26e29.7f0d9","topic":"","payload":{"value":true,"unitid":"2","fc":5,"address":100,"quantity":"1"}},"state":"WRITING"} Serial@/dev/ttyUSB0:9600bit/s unit_id: 1

3 Jan 13:47:50 - [info] [modbus-client:usb-485] Client -> event: connect old: WRITING new: CONNECTED Serial@/dev/ttyUSB0:9600bit/s unit_id: 1

3 Jan 13:47:50 - [error] [modbus-write:teste1] Modbus exception 3

3 Jan 13:47:50 - [info] [modbus-write:teste1] setModbusError: "Modbus exception 3" msg: {"_msgid":"80f26e29.7f0d9","topic":"","payload":{"value":true,"unitid":"2","fc":5,"address":100,"quantity":"1"}}

biancode, 2017-01-03:
_Client -> {"info":"write msg","msg":{"_msgid":"80f26e29.7f0d9","topic":"","payload":{"value":true,"unitid":"2","fc":5,"address":100,"quantity":"1"}},"state":"WRITING"} Serial@/dev/ttyUSB0:9600bit/s unit_id: 1_ that gives Unit Id 2 into modbus-serial  and 1 isn't used until unitid has a value - if your device has Unit Id 2 there could be a problem in modbus-serial

presuntoufu, 2017-01-03:
When i use a modbus-getter, it works properly and i have this response:

3 Jan 14:10:50 - [info] [modbus-client:usb-485] {"loopyLength":0,"commandData":{"msg":{"topic":"76aa9992.f6ea9","payload":{"unitid":"2","fc":3,"address":"43","quantity":"1"}}},"delay":100}

3 Jan 14:10:50 - [info] [modbus-client:usb-485] Client -> {"info":"read msg","msg":{"topic":"76aa9992.f6ea9","payload":{"unitid":"2","fc":3,"address":"43","quantity":"1"}},"state":"READING"} Serial@/dev/ttyUSB0:9600bit/s unit_id: 1

3 Jan 14:10:50 - [info] [modbus-client:usb-485] Client -> event: connect old: READING new: CONNECTED Serial@/dev/ttyUSB0:9600bit/s unit_id: 1

Looking into two responses, the payload in modbus-write starts with value, and payload in modbus-getter starts with unitid. The order matters in payload? 

Another diference in both responses is that in modbus-write, the address in payload is a number, and in modbus-getter is a string. Maybe this is a problem?

biancode, 2017-01-03:
from next version you got more information like

Client -> {"info":"read msg","msg":{"topic":"","payload":{"unitid":"244","fc":3,"address":"0","quantity":"20"}},"unitid":244,"timeout":1000,"state":"READING"} TCP@127.0.0.1:10502 default Unit-Id: 0

biancode, 2017-01-03:
please test v0.11.13

yaacov, 2017-01-03:
some questions:
a. we are using the buffered serial connection ?
b. are we blocking until one read/write interaction is over before we start a new one ?

presuntoufu, 2017-01-03:
biancode, i don't have any modbus hardware here, so i will test tomorrow morning and give you some feedback.

yaacov,
a. Yes, we are using RTU Buffered from node-red-contrib-modbus
b. In this especific test, i'm only read one time, then when i got the response, i write a command.

biancode, 2017-01-03:
@yaacov 
b. the connection node handles the communication based on the response with a queue or without. (config) With the queue it sends the next command after response or error with checking the error. Without queue it is just able to send data if the connection state is connected - a tiny state machine switching between connected, reading, writing

yaacov, 2017-01-03:
@biancode 👍 , when using serial line we have to make sure we use queue, in tcp is less important.

@presuntoufu say he is waiting for read to end before sending write. So it does not sound like overlapping of signals from the client answer and the new request being sent over it.

presuntoufu, 2017-01-04:
@biancode, i've tried v0.11.13 and still got "modbus exception 3". In this version, the default unit id in modbus-client overwrites de unit id in modbus-getter. I just receive a modbus response from my slave (2) when i set default unit id to 2. When default unit id is set to 1, modbus-write and modbus-getter "returns timed out". When default unit id is set to 2, modbus-write returns "modbus exception 3" and modbus-getter returns the expected response.

yaacov, 2017-01-04:
@presuntoufu 
do you have nodejs and npm installed on your raspberry ?
can you try to run some commands on a nodejs console, so we can see some more low level error messages ?

presuntoufu, 2017-01-04:
@yaacov , here in my work i'm using a notebook for tests, and in both (raspberry pi and notebook) i have nodejs and npm installed. 
Yes, i can try the commands, i just don't know how. Can you explain to me?

yaacov, 2017-01-04:
@presuntoufu yes :-)

1.
a. you need to run the node console on the machine connected to the device.
b. you need to install serialport for communicating with tty's - 
`npm install serialport`
c. you need to install the low level modbus library:
`npm install modbus-serial`

2. open a node console:
and write this:
```
// import the modbus library
var ModbusRTU = require("modbus-serial");
var client = new ModbusRTU();

// open connection to a serial port [ try connectRTUBuffered and then connectRTU] see if something changes
client.connectRTU("/dev/ttyUSB0", {baudrate: 9600});

// read
client.setID(1);
client.readInputRegisters(0, 10).then(console.log).catch(console.log)
```

presuntoufu, 2017-01-04:
@yaacov , i've installed the npm packages and run in terminal the command "node", and when i copy-paste the first line of code in terminal, returns "undefined". What i'm doing wrong?

yaacov, 2017-01-04:
"undefined" is great :-) it's ok ( that is what you get on a terminal when a function return void )

presuntoufu, 2017-01-04:
@yaacov ,

> var ModbusRTU = require("modbus-serial");
undefined
> var client = new ModbusRTU();
undefined
> client.connectRTU("/dev/ttyUSB0", {baudrate: 9600});
undefined
> client.setID(1);
undefined
> client.readInputRegisters(0, 10).then(console.log).catch(console.log)
Promise { <pending> }
>

yaacov, 2017-01-04:
ok, 

1. use can use control-C to exit the promise waiting ( if the console stuck on it ... )
2. set the id to your machine
3. read the address you know you have on your machine:
for example:
```
client.setID(2);
client.readHoldingRegisters(43, 1)
```

presuntoufu, 2017-01-04:
@yaacov ,

> var ModbusRTU = require("modbus-serial");
undefined
> var client = new ModbusRTU();
undefined
> client.connectRTU("/dev/ttyUSB0", {baudrate: 9600});
undefined
> client.setID(2);
undefined
> client.readInputRegisters(2, 10).then(console.log).catch(console.log)
Promise { <pending> }
> Data length error, expected 25 got 2
> client.readHoldingRegisters(2, 10).then(console.log).catch(console.log)
Promise { <pending> }
> Data length error, expected 25 got 1

My modbus hardware expects 9600 8-N-2, where can i change it?

I saw this response in node-red, when i'm using unbuffered RTU. In node-red, i receive slave response only when i set to RTU buffered

yaacov, 2017-01-04:
ok, start from the begining but this time use buffered:
( I think the default is 8-N-1 , but I added it below )

```
var ModbusRTU = require("modbus-serial");
var client = new ModbusRTU();
client.connectRTUBuffered("/dev/ttyUSB0", {baudrate:9600, dataBits: 8, parity: 'none', stopBits: 1});

client.readHoldingRegisters(2, 10).then(console.log).catch(console.log)

```

yaacov, 2017-01-04:
I forgot the setid ^^

presuntoufu, 2017-01-04:
> var ModbusRTU = require("modbus-serial");
undefined
> var client = new ModbusRTU();
undefined
> client.connectRTUBuffered("/dev/ttyUSB0", {baudrate:9600, dataBits: 8, parity: 'none', stopBits: 2});
undefined
> client.setID(2);
undefined
> client.readHoldingRegisters(2, 10).then(console.log).catch(console.log)
Promise { <pending> }
> { data: [ 0, 0, 0, 0, 0, 0, 0, 0, 0, 0 ],
  buffer: <Buffer 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00> }

biancode, 2017-01-04:
that is what the node-red node does:

node config for serial should be:  RTU-Buffered /dev/ttyUSB0 baudrate:9600 dataBits: 8 parity: 'none' stopBits: 2

client.readHoldingRegisters(2, 10)  is address 2 and quantity 10

and you should try to use the queue of commands  ;)

yaacov, 2017-01-04:
@biancode yes, but we want to check what happen if you re-set the id after you started using one id,

@presuntoufu can you test:

a. if you set the id, the next read is sent correctly ?
e.g. this set of commands will read from two different machines ?
```
client.setID(2);
client.readHoldingRegisters(2, 10).then(console.log).catch(console.log)
client.setID(3);
client.readHoldingRegisters(2, 10).then(console.log).catch(console.log)
```

b. what happen if you try to write ?
e.g. writing two registers starting from adress 5 , use adress that you know the machine allow writing to.
`client.writeRegisters(5, [0 , 0xffff])`

yaacov, 2017-01-04:
forgot the `.then(console.log).catch(console.log)` after the write ^^

presuntoufu, 2017-01-04:
now i don't have two modbus machines here. How can i write with FC5? I'm trying to use the same address and registers that i've used in node-red, so i need to write to a coil in address 100, value true.

presuntoufu, 2017-01-04:
> var ModbusRTU = require("modbus-serial");
undefined
> var client = new ModbusRTU();
undefined
> client.connectRTUBuffered("/dev/ttyUSB0", {baudrate:9600, dataBits: 8, parity: 'none', stopBits: 2});
undefined
> client.setID(2);
undefined
> client.readHoldingRegisters(2, 19).then(console.log).catch(console.log)
Promise { <pending> }
> { data: [ 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 32, 3, 3, 3, 3, 3, 3, 0 ],
  buffer: <Buffer 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 20 00 03 00 03 00 03 00 03 00 03 00 03 00 00> }
> client.readHoldingRegisters(2, 19).then(console.log).catch(console.log)
Promise { <pending> }
> { data: [ 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 32, 3, 3, 3, 3, 3, 3, 0 ],
  buffer: <Buffer 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 20 00 03 00 03 00 03 00 03 00 03 00 03 00 00> }
> client.setID(3);
undefined
> client.readHoldingRegisters(2, 19).then(console.log).catch(console.log)
Promise { <pending> }
> client.setID(2);
undefined
> client.readHoldingRegisters(2, 19).then(console.log).catch(console.log)
Promise { <pending> }
> { data: [ 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 32, 3, 3, 3, 3, 3, 3, 0 ],
  buffer: <Buffer 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 20 00 03 00 03 00 03 00 03 00 03 00 03 00 00> }

yaacov, 2017-01-04:
a. ok so re-setting the id works :-)

b. now for wrting to a coil [ client.writeCoil(address, state) ]
set the coil and value as your machine expect
e.g.
`client.writeCoil(100, true).then(console.log).catch(console.log) `

p.s.
if you have strange errors try a more low level command -
```
// [2 - unit id, 100 -coil adress, true- value to force]
client.writeFC5(2, 100, true, console.log)
```

presuntoufu, 2017-01-04:
I leave the console running so after the tests that i already sent, i send this two new commands, and the response is:

> client.writeFC5(2, 100, true, console.log)
undefined
> Modbus exception 3
> client.writeCoil(100, true).then(console.log).catch(console.log)
Promise { <pending> }
> Modbus exception 3

yaacov, 2017-01-04:
It means you machine does not allow (*) to set coil no 100 ?

a. are you sure your machine allow this command ?
b. can you check with a different modbus client ( python ... ) if it can set coil 100 on your machine ?

(*) 03/0x03 - Illegal Data Value            
The value contained in the query data field is not an allowable 
value for the server (or slave).

yaacov, 2017-01-04:
@biancode conclusions so far :-)
a. `setId` works
b. It looks like the machine @presuntoufu use does not allow to write values into coil #100.

To check more we need:
1. try to set a coils the _machine manual_  or some other documentation promise that is settable.
2. try to set coil 100 using different tool.

presuntoufu, 2017-01-04:
@yaacov , i re-read the manual of my hardware, and the coil 100 to 107 are all writable, but in first, i have to write to coil 104 first to enable. Now it is working. :-) Thanks a lot for your time, you are very helpful. 

@biancode , in v0.11.13, the problem with default unit id and unit id still happening. i'll downgrade to v0.11.12 to test again. Thanks a lot for your time too. You are very kind person.

biancode, 2017-01-04:
@presuntoufu please try 0.11.14  arrrr  string to integer problems

presuntoufu, 2017-01-04:
I tried  the last version and when i start node-red, i received 5 times this message:

(node) warning: possible EventEmitter memory leak detected. 11 mbinit listeners added. Use emitter.setMaxListeners() to increase limit.
Trace
    at ModbusClientNode.addListener (events.js:239:17)
    at ModbusClientNode.Node.on (/usr/local/lib/node_modules/node-red/red/runtime/nodes/Node.js:80:14)
    at new ModbusWrite (/home/erick/.node-red/node_modules/node-red-contrib-modbus/modbus/modbus-write.js:95:18)
    at createNode (/usr/local/lib/node_modules/node-red/red/runtime/nodes/flows/Flow.js:302:18)
    at Flow.start (/usr/local/lib/node_modules/node-red/red/runtime/nodes/flows/Flow.js:87:35)
    at start (/usr/local/lib/node_modules/node-red/red/runtime/nodes/flows/index.js:277:29)
    at /usr/local/lib/node_modules/node-red/red/runtime/nodes/flows/index.js:132:21
    at tryCatchReject (/usr/local/lib/node_modules/node-red/node_modules/when/lib/makePromise.js:845:30)
    at runContinuation1 (/usr/local/lib/node_modules/node-red/node_modules/when/lib/makePromise.js:804:4)
    at Fulfilled.when (/usr/local/lib/node_modules/node-red/node_modules/when/lib/makePromise.js:592:4)
    at Pending.run (/usr/local/lib/node_modules/node-red/node_modules/when/lib/makePromise.js:483:13)
    at Scheduler._drain (/usr/local/lib/node_modules/node-red/node_modules/when/lib/Scheduler.js:62:19)
    at Scheduler.drain (/usr/local/lib/node_modules/node-red/node_modules/when/lib/Scheduler.js:27:9)
    at nextTickCallbackWith0Args (node.js:436:9)
    at process._tickCallback (node.js:365:13)

The modbus is working correctly, reading and writing to my modbus machine.

closed 2017-01-05 by biancode