biancode/node-red-contrib-modbus #33: Possible Memmory leak

davidoc85, 2017-01-05:
I have seen a similar issue.  How can I disable each flow to check issues?
Currently running:
node-red-dashboard
node-red-contrib-modbus
mqtt nodes
Node-red is running very slow.  Any ideas as to why this might be?  (Can open a new issue for this)
The following output came when running node-red as it mentions that there is a memory leak:

`Welcome to Ubuntu 14.04.5 LTS (GNU/Linux 4.4.0-57-generic x86_64)`
`  System information as of Thu Jan  5 13:01:33 GMT 2017`
`  System load: 0.4                Memory usage: 0%   Processes:       268`
`  Usage of /:  2.4% of 423.47GB   Swap usage:   0%   Users logged in: 0`

`Welcome to Node-RED`
`===================`

`5 Jan 15:04:24 - [info] Node-RED version: v0.15.2`
`5 Jan 15:04:24 - [info] Node.js  version: v4.6.2`
`5 Jan 15:04:24 - [info] Linux 4.4.0-57-generic x64 LE`
`5 Jan 15:04:24 - [info] Loading palette nodes`
`5 Jan 15:04:25 - [info] Dashboard version 2.1.0 started at /ui`
`5 Jan 15:04:25 - [warn] ------------------------------------------------------`
`5 Jan 15:04:25 - [warn] [rpi-gpio] Info : Ignoring Raspberry Pi specific node`
`5 Jan 15:04:25 - [warn] ------------------------------------------------------`
`5 Jan 15:04:25 - [info] Settings file  : /home/pancras6/.node-red/settings.js`
`5 Jan 15:04:25 - [info] User directory : /home/pancras6/.node-red`
`5 Jan 15:04:25 - [info] Flows file     : /home/pancras6/.node-red/flows_ubuntu-server.json`
`5 Jan 15:04:25 - [info] Server now running at http://127.0.0.1:1880/`
`5 Jan 15:04:25 - [info] Starting flows`
`2017-01-05 15:04:25: Connecting to modbustcp slave at 100.120.87.49:502 unit_id: 1`
`(node) warning: possible EventEmitter memory leak detected. 11 mbinit listeners added. Use` `emitter.setMaxListeners() to increase limit.`
`Trace at ModbusClientNode.addListener (events.js:239:17)`
`    at ModbusClientNode.Node.on (/usr/lib/node_modules/node-red/red/runtime/nodes/Node.js:80:14)`
`    at new ModbusWrite (/home/pancras6/.node-red/node_modules/node-red-contrib-modbus/modbus` `/modbus-write.js:95:18)`
    `at createNode (/usr/lib/node_modules/node-red/red/runtime/nodes/flows/Flow.js:302:18)`
    `at Flow.start (/usr/lib/node_modules/node-red/red/runtime/nodes/flows/Flow.js:87:35)`
    `at start (/usr/lib/node_modules/node-red/red/runtime/nodes/flows/index.js:277:29)`
    `at tryCatchReject (/usr/lib/node_modules/node-red/node_modules/when/lib/makePromise.js:845:30)`
    `at runContinuation1 (/usr/lib/node_modules/node-red/node_modules/when/lib/makePromise.js:804:4)`
    `at Fulfilled.when (/usr/lib/node_modules/node-red/node_modules/when/lib/makePromise.js:592:4)`
    `at Pending.run (/usr/lib/node_modules/node-red/node_modules/when/lib/makePromise.js:483:13)`
    `at Scheduler._drain (/usr/lib/node_modules/node-red/node_modules/when/lib/Scheduler.js:62:19)`
    `at Scheduler.drain (/usr/lib/node_modules/node-red/node_modules/when/lib/Scheduler.js:27:9)`
    `at nextTickCallbackWith0Args (node.js:436:9)`
    `at process._tickCallback (node.js:365:13)`
`(node) warning: possible EventEmitter memory leak detected. 11 mbconnected listeners added. Use`  `emitter.setMaxListeners() to increase limit.`
`Trace`
   ` at ModbusClientNode.addListener (events.js:239:17)`
    `at ModbusClientNode.Node.on (/usr/lib/node_modules/node-red/red/runtime/nodes/Node.js:80:14)`
    `at new ModbusWrite (/home/pancras6/.node-red/node_modules/node-red-contrib-modbus/modbus` `/modbus-write.js:96:18)`
    `at createNode (/usr/lib/node_modules/node-red/red/runtime/nodes/flows/Flow.js:302:18)`
    `at Flow.start (/usr/lib/node_modules/node-red/red/runtime/nodes/flows/Flow.js:87:35)`
    `at start (/usr/lib/node_modules/node-red/red/runtime/nodes/flows/index.js:277:29)`
    `at tryCatchReject (/usr/lib/node_modules/node-red/node_modules/when/lib/makePromise.js:845:30)`
    `at runContinuation1 (/usr/lib/node_modules/node-red/node_modules/when/lib/makePromise.js:804:4)`
    `at Fulfilled.when (/usr/lib/node_modules/node-red/node_modules/when/lib/makePromise.js:592:4)`
    `at Pending.run (/usr/lib/node_modules/node-red/node_modules/when/lib/makePromise.js:483:13)`
    `at Scheduler._drain (/usr/lib/node_modules/node-red/node_modules/when/lib/Scheduler.js:62:19)`
    `at Scheduler.drain (/usr/lib/node_modules/node-red/node_modules/when/lib/Scheduler.js:27:9)`
    `at nextTickCallbackWith0Args (node.js:436:9)`
   `at process._tickCallback (node.js:365:13)`
`(node) warning: possible EventEmitter memory leak detected. 11 mberror listeners added. Use` `emitter.setMaxListeners() to increase limit.`
`Trace`
    `at ModbusClientNode.addListener (events.js:239:17)`
    `at ModbusClientNode.Node.on (/usr/lib/node_modules/node-red/red/runtime/nodes/Node.js:80:14)`
    `at new ModbusWrite (/home/pancras6/.node-red/node_modules/node-red-contrib-modbus/modbus` `/modbus-write.js:97:18)`
    `at createNode (/usr/lib/node_modules/node-red/red/runtime/nodes/flows/Flow.js:302:18)`
    `at Flow.start (/usr/lib/node_modules/node-red/red/runtime/nodes/flows/Flow.js:87:35)`
    `at start (/usr/lib/node_modules/node-red/red/runtime/nodes/flows/index.js:277:29)`
    `at tryCatchReject (/usr/lib/node_modules/node-red/node_modules/when/lib/makePromise.js:845:30)`
    `at runContinuation1 (/usr/lib/node_modules/node-red/node_modules/when/lib/makePromise.js:804:4)`
    `at Fulfilled.when (/usr/lib/node_modules/node-red/node_modules/when/lib/makePromise.js:592:4)`
    `at Pending.run (/usr/lib/node_modules/node-red/node_modules/when/lib/makePromise.js:483:13)`
    `at Scheduler._drain (/usr/lib/node_modules/node-red/node_modules/when/lib/Scheduler.js:62:19)`
    `at Scheduler.drain (/usr/lib/node_modules/node-red/node_modules/when/lib/Scheduler.js:27:9)`
    `at nextTickCallbackWith0Args (node.js:436:9)`
    `at process._tickCallback (node.js:365:13)`
`(node) warning: possible EventEmitter memory leak detected. 11 mbclosed listeners added. Use` `emitter.setMaxListeners() to increase limit.`
`Trace`
    `at ModbusClientNode.addListener (events.js:239:17)`
    `at ModbusClientNode.Node.on (/usr/lib/node_modules/node-red/red/runtime/nodes/Node.js:80:14)`
    `at new ModbusWrite (/home/pancras6/.node-red/node_modules/node-red-contrib-modbus/modbus` `/modbus-write.js:98:18)`
    `at createNode (/usr/lib/node_modules/node-red/red/runtime/nodes/flows/Flow.js:302:18)`
    `at Flow.start (/usr/lib/node_modules/node-red/red/runtime/nodes/flows/Flow.js:87:35)`
    `at start (/usr/lib/node_modules/node-red/red/runtime/nodes/flows/index.js:277:29)`
    `at tryCatchReject (/usr/lib/node_modules/node-red/node_modules/when/lib/makePromise.js:845:30)`
    `at runContinuation1 (/usr/lib/node_modules/node-red/node_modules/when/lib/makePromise.js:804:4)`
    `at Fulfilled.when (/usr/lib/node_modules/node-red/node_modules/when/lib/makePromise.js:592:4)`
    `at Pending.run (/usr/lib/node_modules/node-red/node_modules/when/lib/makePromise.js:483:13)`
    `at Scheduler._drain (/usr/lib/node_modules/node-red/node_modules/when/lib/Scheduler.js:62:19)`
    `at Scheduler.drain (/usr/lib/node_modules/node-red/node_modules/when/lib/Scheduler.js:27:9)`
    `at nextTickCallbackWith0Args (node.js:436:9)`
    `at process._tickCallback (node.js:365:13)`
`5 Jan 15:04:25 - [info] Started flows`
`5 Jan 15:04:25 - [info] [mqtt-broker:4aebbc11.d6c5fc] Connected to broker: mqtt://localhost:1883`
`5 Jan 15:04:53 - [error] [modbus-write:807] Data length error, expected 8 got 20`
`5 Jan 15:04:58 - [error] [modbus-write:806] Timed out`
`5 Jan 15:05:23 - [error] [modbus-write:807] Data length error, expected 8 got 20`
`5 Jan 15:05:23 - [error] [modbus-write:802] TypeError: value is out of bounds`
`5 Jan 15:05:28 - [error] [modbus-write:806] Timed out`
`5 Jan 15:05:53 - [error] [modbus-write:807] Data length error, expected 8 got 20`

biancode, 2017-01-05:
No, this is not a memory leak. You should have more than 10 nodes connected to one client. This is just a hint.

biancode, 2017-01-05:
if you think it is slow - do you have more information (your config screen, times, how many nodes or how looks your flow, polling or not, queue or not ...), please

davidoc85, 2017-01-05:
How can I poll different points?  I have got 100+ points writing to a modbus client.  I can send you a complete export.  (see text file)
[test.txt](https://github.com/biancode/node-red-contrib-modbus/files/688175/test.txt)

[image]

davidoc85, 2017-01-05:
Verbose log attached.  Please any advise will be helpful
[verboseLogServer.txt](https://github.com/biancode/node-red-contrib-modbus/files/688195/verboseLogServer.txt)

biancode, 2017-01-05:
please use the flex writer and flex getter to clear your communication - example:

[image]

[{"id":"64722a5b.644db4","type":"mqtt in","z":"dc2fa771.9adf78","name":"Temp","topic":"PR/MON01","qos":"2","broker":"fd8372a9.49b9","x":90,"y":100,"wires":[["8fe5e05b.2f68e"]]},{"id":"8d4294e8.2d3f18","type":"mqtt in","z":"dc2fa771.9adf78","name":"Hum","topic":"PR/MON02","qos":"2","broker":"fd8372a9.49b9","x":90,"y":140,"wires":[["9d8e0a99.f3f978"]]},{"id":"54d6d196.3c3bc","type":"mqtt in","z":"dc2fa771.9adf78","name":"Motion","topic":"PR/MON03","qos":"2","broker":"fd8372a9.49b9","x":90,"y":180,"wires":[["75ac3505.38787c"]]},{"id":"285dffc9.8860e","type":"mqtt in","z":"dc2fa771.9adf78","name":"Lighting Fault","topic":"PR/MON04","qos":"2","broker":"fd8372a9.49b9","x":110,"y":220,"wires":[["ed3b6de1.07f0e"]]},{"id":"8fe5e05b.2f68e","type":"function","z":"dc2fa771.9adf78","name":"Value and Register 260","func":"msg.payload = { value: msg.payload };\nmsg.payload.address = 260;\nreturn msg;","outputs":1,"noerr":0,"x":350,"y":100,"wires":[["9b34967f.cf3e98"]]},{"id":"6c17fe59.d3b8b","type":"modbus-flex-write","z":"dc2fa771.9adf78","name":"","showStatusActivities":false,"server":"b6c66310.49c74","x":830,"y":160,"wires":[[],[]]},{"id":"9b34967f.cf3e98","type":"function","z":"dc2fa771.9adf78","name":"FC6 Unit-Id 1","func":"msg.payload.fc = 6;\nmsg.payload.unitid = 1;\nreturn msg;","outputs":1,"noerr":0,"x":620,"y":160,"wires":[["6c17fe59.d3b8b"]]},{"id":"9d8e0a99.f3f978","type":"function","z":"dc2fa771.9adf78","name":"Value and Register 261","func":"msg.payload = { value: msg.payload };\nmsg.payload.address = 261;\nreturn msg;","outputs":1,"noerr":0,"x":350,"y":140,"wires":[["9b34967f.cf3e98"]]},{"id":"75ac3505.38787c","type":"function","z":"dc2fa771.9adf78","name":"Value and Register 262","func":"msg.payload = { value: msg.payload };\nmsg.payload.address = 262;\nreturn msg;","outputs":1,"noerr":0,"x":350,"y":180,"wires":[["9b34967f.cf3e98"]]},{"id":"ed3b6de1.07f0e","type":"function","z":"dc2fa771.9adf78","name":"Value and Register 269","func":"msg.payload = { value: msg.payload };\nmsg.payload.address = 269;\nreturn msg;","outputs":1,"noerr":0,"x":350,"y":220,"wires":[["9b34967f.cf3e98"]]},{"id":"fd8372a9.49b9","type":"mqtt-broker","z":"","broker":"localhost","port":"1883","clientid":"","usetls":false,"compatmode":true,"keepalive":"60","cleansession":true,"willTopic":"","willQos":"0","willPayload":"","birthTopic":"","birthQos":"0","birthPayload":""},{"id":"b6c66310.49c74","type":"modbus-client","z":"","name":"Local Buffer","clienttype":"tcp","bufferCommands":true,"tcpHost":"127.0.0.1","tcpPort":"10502","serialPort":"/dev/ttyS0","serialType":"RTU","serialBaudrate":"9600","serialDatabits":"8","serialStopbits":"1","serialParity":"none","serialConnectionDelay":"100","unit_id":"","commandDelay":"100","clientTimeout":1000,"reconnectTimeout":5000}]

biancode, 2017-01-05:
[image]

[{"id":"70c9e7a4.e6f068","type":"mqtt in","z":"21218cfe.262824","name":"1L1 kW","topic":"BM1/1/02","qos":"2","broker":"fd8372a9.49b9","x":90,"y":40,"wires":[["301fef3d.4354e"]]},{"id":"d0940559.f6a2f8","type":"mqtt in","z":"21218cfe.262824","name":"1L1 kWh","topic":"BM1/1/03","qos":"2","broker":"fd8372a9.49b9","x":100,"y":80,"wires":[["9c247f37.7be83"]]},{"id":"2ad2fe41.fc15e2","type":"mqtt in","z":"21218cfe.262824","name":"1L2 kW","topic":"BM1/2/02","qos":"2","broker":"fd8372a9.49b9","x":90,"y":120,"wires":[["103f1c2a.595114"]]},{"id":"301fef3d.4354e","type":"function","z":"21218cfe.262824","name":"Value faktor 10 and Register 300","func":"msg.payload = { value: msg.payload * 10 };\nmsg.payload.address = 300;\nreturn msg;","outputs":1,"noerr":0,"x":350,"y":40,"wires":[["12a1062b.5e644a"]]},{"id":"12a1062b.5e644a","type":"function","z":"21218cfe.262824","name":"FC6 Unit-Id 1","func":"msg.payload.fc = 6;\nmsg.payload.unitid = 1;\nreturn msg;","outputs":1,"noerr":0,"x":620,"y":80,"wires":[["998d6523.deec18"]]},{"id":"998d6523.deec18","type":"modbus-flex-write","z":"21218cfe.262824","name":"","showStatusActivities":false,"x":830,"y":80,"wires":[[],[]]},{"id":"9c247f37.7be83","type":"function","z":"21218cfe.262824","name":"Value faktor 10 and Register 301","func":"msg.payload = { value: msg.payload * 10 };\nmsg.payload.address = 301;\nreturn msg;","outputs":1,"noerr":0,"x":350,"y":80,"wires":[["12a1062b.5e644a"]]},{"id":"103f1c2a.595114","type":"function","z":"21218cfe.262824","name":"Value faktor 10 and Register 303","func":"msg.payload = { value: msg.payload * 10 };\nmsg.payload.address = 303;\nreturn msg;","outputs":1,"noerr":0,"x":350,"y":120,"wires":[["12a1062b.5e644a"]]},{"id":"fd8372a9.49b9","type":"mqtt-broker","z":"","broker":"localhost","port":"1883","clientid":"","usetls":false,"compatmode":true,"keepalive":"60","cleansession":true,"willTopic":"","willQos":"0","willPayload":"","birthTopic":"","birthQos":"0","birthPayload":""}]

biancode, 2017-01-05:
[image]

[{"id":"70c9e7a4.e6f068","type":"mqtt in","z":"21218cfe.262824","name":"1L1 kW","topic":"BM1/1/02","qos":"2","broker":"fd8372a9.49b9","x":90,"y":40,"wires":[["301fef3d.4354e"]]},{"id":"d0940559.f6a2f8","type":"mqtt in","z":"21218cfe.262824","name":"1L1 kWh","topic":"BM1/1/03","qos":"2","broker":"fd8372a9.49b9","x":100,"y":80,"wires":[["9c247f37.7be83"]]},{"id":"2ad2fe41.fc15e2","type":"mqtt in","z":"21218cfe.262824","name":"1L2 kW","topic":"BM1/2/02","qos":"2","broker":"fd8372a9.49b9","x":90,"y":120,"wires":[["103f1c2a.595114"]]},{"id":"301fef3d.4354e","type":"function","z":"21218cfe.262824","name":"Value and Register 300","func":"msg.payload = { value: msg.payload};\nmsg.payload.address = 300;\nreturn msg;","outputs":1,"noerr":0,"x":330,"y":40,"wires":[["12a1062b.5e644a"]]},{"id":"12a1062b.5e644a","type":"function","z":"21218cfe.262824","name":"Value faktor 10 FC6 Unit-Id 1","func":"msg.payload.value = msg.payload.value * 10;\nmsg.payload.fc = 6;\nmsg.payload.unitid = 1;\nreturn msg;","outputs":1,"noerr":0,"x":620,"y":80,"wires":[["998d6523.deec18"]]},{"id":"998d6523.deec18","type":"modbus-flex-write","z":"21218cfe.262824","name":"","showStatusActivities":false,"x":870,"y":80,"wires":[[],[]]},{"id":"9c247f37.7be83","type":"function","z":"21218cfe.262824","name":"Value and Register 301","func":"msg.payload = { value: msg.payload };\nmsg.payload.address = 301;\nreturn msg;","outputs":1,"noerr":0,"x":330,"y":80,"wires":[["12a1062b.5e644a"]]},{"id":"103f1c2a.595114","type":"function","z":"21218cfe.262824","name":"Value and Register 303","func":"msg.payload = { value: msg.payload };\nmsg.payload.address = 303;\nreturn msg;","outputs":1,"noerr":0,"x":330,"y":120,"wires":[["12a1062b.5e644a"]]},{"id":"fd8372a9.49b9","type":"mqtt-broker","z":"","broker":"localhost","port":"1883","clientid":"","usetls":false,"compatmode":true,"keepalive":"60","cleansession":true,"willTopic":"","willQos":"0","willPayload":"","birthTopic":"","birthQos":"0","birthPayload":""}]

biancode, 2017-01-05:
If you need parallel communication you need multiple clients LocalBufferServer01 LocalBufferServer02 ... some devices are restricted in their amount of connections - first way try to have less modbus nodes - the communication is the same but you get a better handling and very less CPU usage, too ;)

biancode, 2017-01-05:
warning is gone with v0.11.15

davidoc85, 2017-01-06:
Thanks, I will give it a try and let you know the outcome.

closed 2017-01-08 by biancode