biancode/node-red-contrib-modbus #35: Modbus server

creativeng, 2017-01-05:
Hi 
Im testing the modbus server node
it works fine in output, im writing an FC6 to server and it is immediately available on port 10502

if however i write to port 10502 and then i read from buffer, i have very high delay, in the end data is updated, however with more that 30-45 seconds delay...

any idea about this problem?

Thanks
M

biancode, 2017-01-05:
Please paste your flow and I'll give it a try.

creativeng, 2017-01-06:
it seems ok, we found out the problem is in the debug queue, 
if we clean-up the queue we see the data refreshed ... so it works well thanks..

we see however another issue in the tcp connections management, affecting write and read operations
we expect the connection to stay always on, instead after a write (FC16 single address) it shows closed and then it restart connection again.... this takes some time and subsequent read or write operations are delayed...

we don't see however any specific parameter or reason why connection is closed just after the wright. no errors are reported and data is written correctly

M

biancode, 2017-01-06:
the connection is to stay opened - it toggles between connected - reading/writing in normal workflow

if it is closed on your test flow there must be a close or other Error (not timeout) which is closing the connection and with that the config node reconnect the device

please, give me some logs about from verbose mode of your node-red

node-red -v > yourNameV011XX.log

creativeng, 2017-01-06:
Hi,

i have now tested with a clean test node, hereto attached, and it works well, so o think could a problem due to some other interaction....likely something continuosly writing on tcp...
we will rebuild again the flow and test it again..
up to this level it works well 

in the flow you also find the modbus server test...

thanks

[{"id":"f6f7676e.c5b8c8","type":"inject","z":"128daad1.f36215","name":"","topic":"Central on/off","payload":"[1]","payloadType":"json","repeat":"","crontab":"","once":false,"x":161.50001525878906,"y":591.5999755859375,"wires":[["ceb78de5.84d56","6ae92a08.f5fa24"]]},{"id":"f64c1184.48345","type":"modbus-server","z":"128daad1.f36215","name":"TCP-Server-Service","logEnabled":true,"serverPort":10502,"responseDelay":100,"delayUnit":"ms","coilsBufferSize":1024,"holdingBufferSize":1024,"inputBufferSize":1024,"x":657.5000762939453,"y":883.5000305175781,"wires":[[],[],[]]},{"id":"f31b23a5.a72b9","type":"modbus-read","z":"128daad1.f36215","name":"Internal read","showStatusActivities":false,"unitid":"1","dataType":"HoldingRegister","adr":"2","quantity":"1","rate":"1","rateUnit":"s","server":"262a38f6.1f62a8","x":367.5000457763672,"y":724.1999603271483,"wires":[[],[]]},{"id":"ffbdebe5.130708","type":"modbus-write","z":"128daad1.f36215","name":"Internal write","showStatusActivities":true,"unitid":"","dataType":"HoldingRegister","adr":"1","quantity":"1","server":"262a38f6.1f62a8","x":370.50006103515625,"y":814.9997863769531,"wires":[["f64c1184.48345"],[]]},{"id":"7220a721.e2e0f8","type":"inject","z":"128daad1.f36215","name":"","topic":"Central on/off","payload":"[0]","payloadType":"json","repeat":"","crontab":"","once":false,"x":163.1999969482422,"y":642.2000122070312,"wires":[["ceb78de5.84d56","6ae92a08.f5fa24"]]},{"id":"be327f8e.de0ba","type":"inject","z":"128daad1.f36215","name":"test server","topic":"test","payload":"50","payloadType":"num","repeat":"","crontab":"","once":false,"x":326.5000457763672,"y":931.5997619628906,"wires":[["ffbdebe5.130708"]]},{"id":"ceb78de5.84d56","type":"debug","z":"128daad1.f36215","name":"Test Debug Out1","active":true,"console":"false","complete":"true","x":690.4999847412109,"y":643.8000030517578,"wires":[]},{"id":"6ae92a08.f5fa24","type":"modbus-write","z":"128daad1.f36215","name":"Lamp on/off","showStatusActivities":false,"unitid":"","dataType":"MHoldingRegisters","adr":"181","quantity":"1","server":"83940f7.2d620f","x":666.4999847412109,"y":510.40000305175784,"wires":[[],[]]},{"id":"262a38f6.1f62a8","type":"modbus-client","z":"","name":"Local Buffer","clienttype":"tcp","bufferCommands":true,"tcpHost":"127.0.0.1","tcpPort":"10502","serialPort":"/dev/ttyS0","serialType":"RTU","serialBaudrate":"9600","serialDatabits":"8","serialStopbits":"1","serialParity":"none","serialConnectionDelay":"100","unit_id":"","commandDelay":"100","clientTimeout":"1000","reconnectTimeout":"5000"},{"id":"83940f7.2d620f","type":"modbus-client","z":"","name":"DOMATM090","clienttype":"tcp","bufferCommands":false,"tcpHost":"172.21.10.7","tcpPort":"502","serialPort":"/dev/ttyS0","serialType":"RTU","serialBaudrate":"9600","serialDatabits":"8","serialStopbits":"1","serialParity":"none","serialConnectionDelay":"100","unit_id":"1","commandDelay":"100","clientTimeout":"1000","reconnectTimeout":"5000"}]

biancode, 2017-01-08:
hint: servers input is just to get its Buffers on Output

[image]

[{"id":"e0fb7d52.a4f7","type":"tab","label":"Flow 1"},{"id":"49b3394a.8e00d8","type":"inject","z":"e0fb7d52.a4f7","name":"","topic":"Central on/off","payload":"[1]","payloadType":"json","repeat":"","crontab":"","once":false,"x":240,"y":100,"wires":[["a1c156f6.3b0148"]]},{"id":"1eac7572.77f8bb","type":"modbus-server","z":"e0fb7d52.a4f7","name":"Modbus VServer 11502","logEnabled":true,"serverPort":"11502","responseDelay":100,"delayUnit":"ms","coilsBufferSize":1024,"holdingBufferSize":1024,"inputBufferSize":1024,"x":490,"y":380,"wires":[["c6ada80d.445a98"],["c6ada80d.445a98"],["c6ada80d.445a98"]]},{"id":"2bb02cd8.b2c024","type":"modbus-read","z":"e0fb7d52.a4f7","name":"Read VServer Adr 1","showStatusActivities":false,"unitid":"1","dataType":"HoldingRegister","adr":"2","quantity":"1","rate":"1","rateUnit":"s","server":"a600569b.bbcd38","x":470,"y":280,"wires":[["19c371bb.47fe8e"],["dda41787.655758"]]},{"id":"7053f249.ae83ec","type":"modbus-write","z":"e0fb7d52.a4f7","name":"Write VServer Adr 1","showStatusActivities":true,"unitid":"","dataType":"HoldingRegister","adr":"1","quantity":"1","server":"a600569b.bbcd38","x":470,"y":220,"wires":[["19c371bb.47fe8e"],["679f70a6.a162f"]]},{"id":"d7d5369c.184a78","type":"inject","z":"e0fb7d52.a4f7","name":"","topic":"Central on/off","payload":"[0]","payloadType":"json","repeat":"","crontab":"","once":false,"x":241.69998168945312,"y":150.60003662109375,"wires":[["a1c156f6.3b0148"]]},{"id":"e0c8acef.dd665","type":"inject","z":"e0fb7d52.a4f7","name":"test server","topic":"test","payload":"50","payloadType":"num","repeat":"","crontab":"","once":false,"x":260,"y":220,"wires":[["7053f249.ae83ec"]]},{"id":"a1c156f6.3b0148","type":"modbus-write","z":"e0fb7d52.a4f7","name":"Lamp on/off","showStatusActivities":false,"unitid":"","dataType":"MHoldingRegisters","adr":"181","quantity":"1","server":"9846c3e4.bd598","x":450,"y":120,"wires":[["3197da84.d71856"],["6f2009f3.a1fd98"]]},{"id":"91ca1997.227a58","type":"inject","z":"e0fb7d52.a4f7","name":"Read all Buffers","topic":"","payload":"","payloadType":"date","repeat":"","crontab":"","once":false,"x":260,"y":380,"wires":[["1eac7572.77f8bb"]]},{"id":"c6ada80d.445a98","type":"debug","z":"e0fb7d52.a4f7","name":"All Buffers","active":true,"console":"false","complete":"true","x":700,"y":380,"wires":[]},{"id":"679f70a6.a162f","type":"modbus-response","z":"e0fb7d52.a4f7","name":"","registerShowMax":20,"x":710,"y":220,"wires":[]},{"id":"dda41787.655758","type":"modbus-response","z":"e0fb7d52.a4f7","name":"","registerShowMax":20,"x":710,"y":280,"wires":[]},{"id":"3197da84.d71856","type":"debug","z":"e0fb7d52.a4f7","name":"Debug","active":true,"console":"false","complete":"payload","x":650,"y":60,"wires":[]},{"id":"6f2009f3.a1fd98","type":"modbus-response","z":"e0fb7d52.a4f7","name":"","registerShowMax":20,"x":690,"y":100,"wires":[]},{"id":"19c371bb.47fe8e","type":"debug","z":"e0fb7d52.a4f7","name":"Debug VServer Commands","active":true,"console":"false","complete":"true","x":740,"y":180,"wires":[]},{"id":"a600569b.bbcd38","type":"modbus-client","z":"","name":"Local VServer","clienttype":"tcp","bufferCommands":true,"stateLogEnabled":false,"tcpHost":"127.0.0.1","tcpPort":"11502","serialPort":"/dev/ttyS0","serialType":"RTU","serialBaudrate":"9600","serialDatabits":"8","serialStopbits":"1","serialParity":"none","serialConnectionDelay":"100","unit_id":"","commandDelay":"50","clientTimeout":"500","reconnectTimeout":"2000"},{"id":"9846c3e4.bd598","type":"modbus-client","z":"","name":"DOMATM090","clienttype":"tcp","bufferCommands":false,"tcpHost":"172.21.10.7","tcpPort":"502","serialPort":"/dev/ttyS0","serialType":"RTU","serialBaudrate":"9600","serialDatabits":"8","serialStopbits":"1","serialParity":"none","serialConnectionDelay":"100","unit_id":1,"commandDelay":100,"clientTimeout":1000,"reconnectTimeout":5000}]

biancode, 2017-01-08:
https://www.npmjs.com/package/node-red-node-data-generator

[image]

[{"id":"49b3394a.8e00d8","type":"inject","z":"e0fb7d52.a4f7","name":"","topic":"Central on/off","payload":"[1]","payloadType":"json","repeat":"","crontab":"","once":false,"x":380,"y":140,"wires":[["a1c156f6.3b0148"]]},{"id":"1eac7572.77f8bb","type":"modbus-server","z":"e0fb7d52.a4f7","name":"Modbus VServer 11502","logEnabled":true,"serverPort":"11502","responseDelay":100,"delayUnit":"ms","coilsBufferSize":1024,"holdingBufferSize":1024,"inputBufferSize":1024,"x":630,"y":420,"wires":[["c6ada80d.445a98"],["c6ada80d.445a98"],["c6ada80d.445a98"]]},{"id":"2bb02cd8.b2c024","type":"modbus-read","z":"e0fb7d52.a4f7","name":"Read VServer Adr 1","showStatusActivities":true,"unitid":"1","dataType":"HoldingRegister","adr":"1","quantity":"3","rate":"1","rateUnit":"s","server":"a600569b.bbcd38","x":610,"y":320,"wires":[["19c371bb.47fe8e"],["dda41787.655758"]]},{"id":"7053f249.ae83ec","type":"modbus-write","z":"e0fb7d52.a4f7","name":"Write VServer Adr 1","showStatusActivities":false,"unitid":"","dataType":"MHoldingRegisters","adr":"1","quantity":"3","server":"a600569b.bbcd38","x":610,"y":260,"wires":[["19c371bb.47fe8e"],["679f70a6.a162f"]]},{"id":"d7d5369c.184a78","type":"inject","z":"e0fb7d52.a4f7","name":"","topic":"Central on/off","payload":"[0]","payloadType":"json","repeat":"","crontab":"","once":false,"x":381.6999816894531,"y":190.60003662109375,"wires":[["a1c156f6.3b0148"]]},{"id":"e0c8acef.dd665","type":"inject","z":"e0fb7d52.a4f7","name":"test server","topic":"test","payload":"50","payloadType":"num","repeat":"1","crontab":"","once":false,"x":230,"y":260,"wires":[["913d724d.e371c"]]},{"id":"a1c156f6.3b0148","type":"modbus-write","z":"e0fb7d52.a4f7","name":"Lamp on/off","showStatusActivities":false,"unitid":"","dataType":"MHoldingRegisters","adr":"181","quantity":"1","server":"9846c3e4.bd598","x":590,"y":160,"wires":[["3197da84.d71856"],["6f2009f3.a1fd98"]]},{"id":"91ca1997.227a58","type":"inject","z":"e0fb7d52.a4f7","name":"Read all Buffers","topic":"","payload":"","payloadType":"date","repeat":"","crontab":"","once":false,"x":400,"y":420,"wires":[["1eac7572.77f8bb"]]},{"id":"c6ada80d.445a98","type":"debug","z":"e0fb7d52.a4f7","name":"All Buffers","active":true,"console":"false","complete":"true","x":840,"y":420,"wires":[]},{"id":"679f70a6.a162f","type":"modbus-response","z":"e0fb7d52.a4f7","name":"","registerShowMax":20,"x":850,"y":260,"wires":[]},{"id":"dda41787.655758","type":"modbus-response","z":"e0fb7d52.a4f7","name":"","registerShowMax":20,"x":850,"y":320,"wires":[]},{"id":"3197da84.d71856","type":"debug","z":"e0fb7d52.a4f7","name":"Debug","active":false,"console":"false","complete":"payload","x":790,"y":100,"wires":[]},{"id":"6f2009f3.a1fd98","type":"modbus-response","z":"e0fb7d52.a4f7","name":"","registerShowMax":20,"x":830,"y":140,"wires":[]},{"id":"19c371bb.47fe8e","type":"debug","z":"e0fb7d52.a4f7","name":"Debug VServer Commands","active":false,"console":"false","complete":"true","x":880,"y":220,"wires":[]},{"id":"332850ab.e761b","type":"catch","z":"e0fb7d52.a4f7","name":"","scope":null,"x":280,"y":100,"wires":[["e31e80d3.f2524"]]},{"id":"e31e80d3.f2524","type":"debug","z":"e0fb7d52.a4f7","name":"","active":true,"console":"false","complete":"true","x":440,"y":100,"wires":[]},{"id":"913d724d.e371c","type":"data-generator","z":"e0fb7d52.a4f7","name":"","field":"payload","fieldType":"msg","syntax":"json","template":"[{{int 1 65000}}, {{int 1 65000}}, {{int 1 65000}}]","x":400,"y":260,"wires":[["7053f249.ae83ec"]]},{"id":"a600569b.bbcd38","type":"modbus-client","z":"","name":"Local VServer","clienttype":"tcp","bufferCommands":true,"stateLogEnabled":false,"tcpHost":"127.0.0.1","tcpPort":"11502","serialPort":"/dev/ttyS0","serialType":"RTU","serialBaudrate":"9600","serialDatabits":"8","serialStopbits":"1","serialParity":"none","serialConnectionDelay":"100","unit_id":"","commandDelay":"100","clientTimeout":"1000","reconnectTimeout":"2000"},{"id":"9846c3e4.bd598","type":"modbus-client","z":"","name":"DOMATM090","clienttype":"tcp","bufferCommands":false,"tcpHost":"172.21.10.7","tcpPort":"502","serialPort":"/dev/ttyS0","serialType":"RTU","serialBaudrate":"9600","serialDatabits":"8","serialStopbits":"1","serialParity":"none","serialConnectionDelay":"100","unit_id":1,"commandDelay":100,"clientTimeout":1000,"reconnectTimeout":5000}]

biancode, 2017-01-08:
[image]

[{"id":"a3c812c6.51c38","type":"modbus-server","z":"7bcd06b6.855108","name":"","logEnabled":false,"serverPort":10502,"responseDelay":100,"delayUnit":"ms","coilsBufferSize":1024,"holdingBufferSize":1024,"inputBufferSize":1024,"x":200,"y":60,"wires":[[],[],[]]},{"id":"3ce1c93.0f57636","type":"modbus-flex-write","z":"7bcd06b6.855108","name":"","showStatusActivities":false,"server":"1b75976b.782c69","x":770,"y":180,"wires":[[],["ce72b528.82bc08"]]},{"id":"ce72b528.82bc08","type":"modbus-response","z":"7bcd06b6.855108","name":"","registerShowMax":20,"x":990,"y":180,"wires":[]},{"id":"23813dce.af2572","type":"modbus-response","z":"7bcd06b6.855108","name":"","registerShowMax":"3","x":530,"y":300,"wires":[]},{"id":"437cd3de.404ecc","type":"inject","z":"7bcd06b6.855108","name":"","topic":"","payload":"","payloadType":"date","repeat":"1","crontab":"","once":false,"x":110,"y":180,"wires":[["3d03bbf2.729664"]]},{"id":"7e0f7cc6.84ed54","type":"inject","z":"7bcd06b6.855108","name":"","topic":"","payload":"","payloadType":"date","repeat":"1","crontab":"","once":false,"x":110,"y":300,"wires":[["b0baf595.42b928"]]},{"id":"3d03bbf2.729664","type":"data-generator","z":"7bcd06b6.855108","name":"","field":"payload","fieldType":"msg","syntax":"json","template":"[{{int 1 65000}}, {{int 1 65000}}, {{int 1 65000}}, {{int 1 65000}}]","x":280,"y":180,"wires":[["3bfcd2f9.a874ce"]]},{"id":"3bfcd2f9.a874ce","type":"function","z":"7bcd06b6.855108","name":"FC16 - Unit-ID 1 - 0-4 Register","func":"msg.payload = { \n    value: msg.payload,\n    fc: 16,\n    unitid: 1,\n    address: 0,\n    quantity: 4\n};\nreturn msg;","outputs":1,"noerr":0,"x":510,"y":180,"wires":[["3ce1c93.0f57636"]]},{"id":"b0baf595.42b928","type":"modbus-getter","z":"7bcd06b6.855108","name":"","showStatusActivities":false,"unitid":"1","dataType":"HoldingRegister","adr":"0","quantity":"4","server":"56e27c81.9dd394","x":320,"y":300,"wires":[[],["23813dce.af2572"],[]]},{"id":"cdc9d4b3.b14fd8","type":"catch","z":"7bcd06b6.855108","name":"","scope":null,"x":440,"y":80,"wires":[["62bdac69.c8db24"]]},{"id":"62bdac69.c8db24","type":"debug","z":"7bcd06b6.855108","name":"","active":true,"console":"false","complete":"false","x":730,"y":80,"wires":[]},{"id":"41477517.ad577c","type":"modbus-flex-getter","z":"7bcd06b6.855108","name":"","showStatusActivities":false,"server":"56e27c81.9dd394","x":640,"y":380,"wires":[[],["a4f99d53.f0cde"]]},{"id":"878bc2aa.3b9eb","type":"inject","z":"7bcd06b6.855108","name":"","topic":"","payload":"","payloadType":"date","repeat":"1","crontab":"","once":false,"x":110,"y":380,"wires":[["5714739e.9411ec"]]},{"id":"5714739e.9411ec","type":"function","z":"7bcd06b6.855108","name":"FC3 - Unit-ID 1 - 0-4 Register","func":"msg.payload = { \n    value: msg.payload,\n    fc: 3,\n    unitid: 1,\n    address: 0,\n    quantity: 4 \n};\nreturn msg;","outputs":1,"noerr":0,"x":350,"y":380,"wires":[["41477517.ad577c"]]},{"id":"a4f99d53.f0cde","type":"modbus-response","z":"7bcd06b6.855108","name":"","registerShowMax":20,"x":890,"y":380,"wires":[]},{"id":"1b75976b.782c69","type":"modbus-client","z":"","name":"Write VServer","clienttype":"tcp","bufferCommands":true,"stateLogEnabled":false,"tcpHost":"127.0.0.1","tcpPort":"10502","serialPort":"/dev/ttyUSB","serialType":"RTU-BUFFERD","serialBaudrate":"9600","serialDatabits":"8","serialStopbits":"1","serialParity":"none","serialConnectionDelay":"100","unit_id":1,"commandDelay":1,"clientTimeout":1000,"reconnectTimeout":2000},{"id":"56e27c81.9dd394","type":"modbus-client","z":"","name":"Read VServer","clienttype":"tcp","bufferCommands":true,"stateLogEnabled":false,"tcpHost":"127.0.0.1","tcpPort":"10502","serialPort":"/dev/ttyUSB","serialType":"RTU-BUFFERD","serialBaudrate":"9600","serialDatabits":"8","serialStopbits":"1","serialParity":"none","serialConnectionDelay":"100","unit_id":"1","commandDelay":"1","clientTimeout":"1000","reconnectTimeout":"2000"}]

creativeng, 2017-01-08:
thanks for the hints, 
this way works better....

also the flex getter/write are very useful..

biancode, 2017-01-08:
[YouTube Playlist v0.11](https://www.youtube.com/playlist?list=PLdcYFhtEB7okPHU1-NUYTeKFNbJOW_-gH)

closed 2017-01-08 by biancode