biancode/node-red-contrib-modbus #36: Float 32bit points modbus write/read

davidoc85, 2017-01-06:
At the moment we have been sending integer strings that consist of 16 bit, however when larger numbers are required these are usually set as a floating points with a 32bit length.  Is it possible to write a 32 bit float to a modbusTCP slave or read a 32bit float point and how is the best way to do it?

biancode, 2017-01-06:
4Bit -> read 16 and split it into four 4Bit
8Bit -> read 16 and split it into two 8Bit
16Bit -> just read
32Bit -> read two 16Bit and connect them
64Bit -  read four 16Bit and connect them

biancode, 2017-01-08:
[image]

[{"id":"76233bf0.471f84","type":"inject","z":"edb24723.cd7c08","name":"","topic":"","payload":"[3,59392,200,3456]","payloadType":"json","repeat":"","crontab":"","once":false,"x":430,"y":80,"wires":[["7597366f.2e0c08"]]},{"id":"70e28431.a8e1dc","type":"function","z":"edb24723.cd7c08","name":"32 to 16","func":"msg = {payload: {'high': msg.payload >> 16, 'low': msg.payload & 0xFFFF}}\nreturn msg;","outputs":1,"noerr":0,"x":660,"y":140,"wires":[["331322e5.a9b0fe"]]},{"id":"b0236c69.57688","type":"inject","z":"edb24723.cd7c08","name":"","topic":"","payload":"256000","payloadType":"num","repeat":"","crontab":"","once":false,"x":470,"y":140,"wires":[["70e28431.a8e1dc"]]},{"id":"7597366f.2e0c08","type":"function","z":"edb24723.cd7c08","name":"16 to 32","func":"msg = { payload: msg.payload[0] << 16 | msg.payload[1] }\nreturn msg;","outputs":1,"noerr":0,"x":660,"y":80,"wires":[["331322e5.a9b0fe"]]},{"id":"331322e5.a9b0fe","type":"debug","z":"edb24723.cd7c08","name":"","active":true,"console":"false","complete":"false","x":830,"y":120,"wires":[]}]

biancode, 2017-01-08:
http://stackoverflow.com/questions/4414077/read-write-bytes-of-float-in-js

closed 2017-01-14 by biancode