biancode/node-red-contrib-modbus #39: Can't install serialport-v4.0.7 as required

Killerloop2, 2017-01-09:
Hello, I can't seem to get these nodes to work.

I'm trying to write a FC6 request over ttyAMA0, when I try to install the latest version I get:

"node-pre-gyp ERR! Tried to download(404): https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v46-linux-arm.tar.gz 

node-pre-gyp ERR! Pre-built binaries not found for serialport@4.0.7 and node@4.7.0 (node-v46 ABI) (falling back to source compile with node-gyp)"

If I try to use the nodes anyway (and they appear in the browser) I get warnings in node-red consoles and nothing works.

Node-red 0.15.3
NPM 2.15.11
Node v4.7.0
NodeJS v4.7.2

biancode, 2017-01-09:
404 error page not found - is it possible to download it via wget?

Killerloop2, 2017-01-09:
wget replies with "404 - page not found". The same happens when trying "npm install serialport"

This is weird since the serialport nodes provided with node-red actually work. May I ask what version of the serialport and node-serialport modules you use during tests?

biancode, 2017-01-09:
https://www.npmjs.com/package/serialport

closed 2017-01-10 by Killerloop2

biancode, 2017-01-14:
I'm using ubuntu mate 16.04 LTS on PI2 https://www.digitalocean.com/community/tutorials/how-to-install-node-js-on-ubuntu-16-04

biancode, 2017-01-14:
done on ubuntu mate 16.04 LTS - node was 6.9.4, but we should use >=4.7 and <=6.9.2

sudo npm cache clean -f
sudo npm install -g n
sudo n 6.9.2
sudo npm install -g serialport@4.0.7 --unsafe-perm --build-from-source
sudo npm install -g node-red-contrib-modbus

reboot and check the node -v shows 6.9.2

biancode, 2017-01-15:
notice https://nodered.org/docs/getting-started/installation