biancode/node-red-contrib-modbus #40: node-red exiting !!!

FrankD11, 2017-01-11:
Hi, I'm testing communications with an arduino and the flow is crashing node-red !

I'm running node-red 15.3 on node.js 4.7 on Windows 10 x64
the link is through an rs 485 usb adapter through com12

Before It started crashing node-red I was getting modbus errors only then node-red started crashing
with this error

> TypeError: node.messageAllowedStates.includes is not a function
    at ModbusClientNode.node.dequeueCommand [as _onTimeout] (D:\Users\removed\AppData\Roaming\npm\node_modules\node-red-contrib-modbus\modbus\modbus-client.js:112:38)
    at Timer.listOnTimeout (timers.js:92:15)

I have tested with "node-red-contrib-serial-modbus v. 0.0.10" and it works
Also I have success with modpoll 3.4

The flow is very simple as below

`[{"id":"f5919f2c.c96fd","type":"modbus-read","z":"c90f3ef6.7168b","name":"","showStatusActivities":false,"unitid":"","dataType":"HoldingRegister","adr":"40000","quantity":"1","rate":"1","rateUnit":"s","server":"8955670c.8ca758","x":166,"y":150,"wires":[["fe29a200.59355"],["f5c19830.ea7128"]]},{"id":"8955670c.8ca758","type":"modbus-client","z":"","name":"","clienttype":"serial","bufferCommands":true,"stateLogEnabled":false,"tcpHost":"127.0.0.1","tcpPort":"502","serialPort":"COM12","serialType":"RTU","serialBaudrate":"9600","serialDatabits":"8","serialStopbits":"1","serialParity":"none","serialConnectionDelay":"100","unit_id":"7","commandDelay":"1","clientTimeout":"1000","reconnectTimeout":"2000"}]`

What would you suggest I do.

Best Regards

Frank

biancode, 2017-01-11:
https://developer.mozilla.org/en/docs/Web/JavaScript/Reference/Global_Objects/Array/includes

may an Browser problem ? Your JS engine should be to old.

biancode, 2017-01-11:
I changed to the old function indexOf in v0.11.20 - give it a try ;)

FrankD11, 2017-01-12:
Hi Klaus,

Thank's for the quick response.

I had to redo my install an then got it working with node 6.9.3 latest and node-red 0.16

Also many thanks for this awesome contribution.

Best regards

closed 2017-01-12 by FrankD11